solokeys/solo1 #489: Build problems for the solo application.

xnfra, 2020-11-21:
I'm trying to build the Solo as a standalone application but it keeps failing. Here is the build log.
[build-error for solo.txt](https://github.com/solokeys/solo/files/5576699/build-error.for.solo.txt)

enrikb, 2020-11-21:
I can reproduce the issue on Ubuntu 20.10 (gcc 10). As I only updated recently and did not see this before, I tried to build using gcc 9, and it worked. (`CC=gcc-9 make`)

enrikb, 2020-11-21:
Can you please check if #490 fixes the issue for you?

xnfra, 2020-11-21:
> Can you please check if #490 fixes the issue for you?

Cloned the repo and the submodules.
`git clone --recurse-submodules https://github.com/solokeys/solo`

Fetched the pull request and checkout.
`git fetch origin pull/490/head:490`
`git checkout 490`

Running `make all` successfully builds the `main` application.
[build-complete.txt](https://github.com/solokeys/solo/files/5578424/build-complete.txt)

All seems well to merge to main. Thanks!

closed 2021-07-18 by xnfra