solokeys/solo1 #491: Applying Solokeys on LUKS

kagamineryuki, 2020-12-01:
Hi, I recently bought this Solokeys and want to use the solokeys as second factor for my LUKS encrypted HDD.
i search for guide but what i found is for yubikey.

is it possible to apply this key to LUKS and if so, how to do it ?

szszszsz, 2020-12-01:
See https://github.com/shimunn/fido2luks

kagamineryuki, 2020-12-01:
> 
> 
> See https://github.com/shimunn/fido2luks

nice, thx for the suggestion. i'll try it out.

closed 2020-12-02 by kagamineryuki

nickray, 2020-12-08:
Also see https://github.com/solokeys/solo-python/pull/105

saravanan30erd, 2020-12-12:
@kagamineryuki 

https://github.com/saravanan30erd/solokey-full-disk-encryption

It supports only Arch linux, but I can add support for other linux distros as well. which linux distro you are using?

kagamineryuki, 2020-12-12:
> @kagamineryuki
> 
> https://github.com/saravanan30erd/solokey-full-disk-encryption
> 
> It supports only Arch linux, but I can add support for other linux distros as well. which linux distro you are using?

I'm currently using Fedora and Debian. Is it specifically solokeys or any FIDO2/U2F devices?

saravanan30erd, 2020-12-12:
Is it specifically solokeys or any FIDO2/U2F devices?

It is created specifically for solokeys.