solokeys/solo1 #492: Programming Problem with my own SoloKey

lucillle, 2020-12-07:
Hi,
I made my own solo key hardware exactly from the schematic and BOM, at the first plug DFU mode is available, according to the [documentation](https://docs.solokeys.dev/programming/#programming-a-solo-that-hasnt-been-programmed) I flashed the official  bundle-hacker-4.0.0.hex and then:

```
$ solo program aux leave-dfu
Rewriting option bytes...
Warning: OSError with write_page
Please powercycle the device (pull out, plug in again)

```
I did powercycle but nothing happened! no detection, no log, no led blink...
seems solo key is broken! I revive it by full erase with jtag and stm32cube tool.

* I saw issue #206
* I did flash my local build, or only bootloader and other scenarios

Best Regards.

enrikb, 2020-12-07:
Did you use `solo` for the actual DFU mode programming, too? If so, can you check solokeys/solo-python#98?

DrArmaggedon, 2020-12-07:
Hi guys, I have the same problem. The thing is LED is working but there is no USB activity. I am a little bit old school So I use STM32CubeProgrammer to directly write bundle-hacker-4.0.0.hex and change the Option Bytes. 
Some other notes:

- Hardware works fine, I have tried to program and run various tests and it works fine.

- LED blinks and work (So the bootloader works)

- USB works fine cause I have successfully program and verified over DFU using STM32CubeProgrammer.

Any suggestion? Is there any way to read the debug log?