solokeys/solo1 #494: Error fetching version name from solokeys/solo repository!

ncm, 2020-12-18:
When I run `solo key update`, it just says "Error fetching version name from solokeys/solo repository!".  It is trying to look at 
`fetch_url=f'https://raw.githubusercontent.com/solokeys/solo/master/STABLE_VERSION'`

Running `requests.get(fetch_url)` interactively, I see:
```
  ssl.SSLError: [SSL: TLSV1_ALERT_PROTOCOL_VERSION] tlsv1 alert protocol version (_ssl.c:1123)
```
When doing `pip3 install solo`, I had to edit my `/etc/ssl/openssl.conf` to change `s/MinProtocol = TLSv1.3/MinProtocol = TLSv1.0/` temporarily, in order to get the solo package downloaded. But that did not suffice for `solo key update`.

(FWIW, the web updater did work to update my Somu.)

As a side issue, maybe github could be persuaded to support a current TLS version on `raw.githubusercontent.com`?

nickray, 2020-12-29:
This seems to me to be a solo-python issue, not a solo issue. Moreover, this seems like an overly restrictive setting on your machine? It's not like TLSv1.2 is going to disappear anytime soon.

closed 2020-12-29 by nickray

ncm, 2020-12-30:
TLS 1.2 is still pretty common, but that does not make using and demanding TLS 1.0 good policy -- particularly for a security-related product. And, as I noted, even permitting TLS 1.0 did not suffice for `solo key update`: that *completely failed*.

Closing the issue doesn't mean things work. As it is, without the web updater, my Solo device would not have been updated at all.

nickray, 2020-12-30:
To expand: Testing with `curl --tlsv1.2 $URL` and `curl --tlsv1.3 $URL` indicates that GitHub supports TLS v1.2 but indeed not TLS v1.3, which is a surprise to me, but hardly qualifies as a security issue. You can try to petition GitHub to upgrade of course. But pragmatically, could you not simply set your MinProtocol to v1.2? What OS/distribution do you use? Is this setting in openssl.cnf a default or your choice?

I don't understand why you argue against TLS v1.0; this seems like a strawman?