solokeys/solo1 #501: FIDO2 stack on nRF52840 DK + BLE support?

coco21, 2021-01-21:
Hello,

I was wondering how easily it is to port and make run your FIDO2 stack on a nRF52840 DK and that communication goes over BLE rather than USB. Studied quite deeply your porting doc pages and understood main interfaces between your FIDO2 stack and the communication channels is (today) USB HID abstraction layer. I was thinking about running your FIDO2 stack on a Nordic native Zephyr-based OS as you can find in their sample code and use their e.g. S313/340 based softdevice BLE stacks (https://www.thisisant.com/developer/components/nrf52832/).

My question: how hard, difficult would it be to achieve that? Does this make sense in your eyes?
I also am testing and running OpenSK FIDO2 stack but that also is missing BLE stack support as of today with a connection to the FIDO2 stack.

Thank you for a quick answer, I do this for an evaluation phase and need to take some decisions.

nickray, 2021-01-21:
Hi @coco21, this is two questions in one :)

On the one hand, portability as such. The existing v1 key C code did run on other platforms in the past, and has been adopted by other projects. The idea would be to add a new target under `targets/`. Whether it's a good idea, especially integrating into an embedded OS, I do not know. For the v2 key Rust code (coming very soon!), I intend to port it to the nRF52840 Dongle ASAP, as a proof of concept of that code base's "modularity and portability". However, this would be in that code base's frameworks (https://rtic.rs/ for scheduling, our own Trussed framework for cryptography and secure storage). The blocker for that is USB support (https://github.com/nrf-rs/nrf-hal/pull/133, https://github.com/nrf-rs/nrf-hal/pull/199, https://github.com/nrf-rs/nrf-hal/pull/199).

The other is Bluetooth. The open question is how sensible that is for a security key in general. As for specifics, on the Rust side, I don't have any experience with BLE, so I can't say much unfortunately.

If you'd like to discuss, I'm @nickray on Keybase, or you can hop into https://matrix.to/#/#solokeys-dev:matrix.org

coco21, 2021-01-22:
Hey @nickray thanks so much, that was quick!
I have good hopes to port Solo FIDO2 stack onto another platform, and it's C, that one we master. Liked the idea of weak definitions and HID as the interface between the FIDO2 stack and the USB stack - was thinking swapping out USB by BLE stack at HID level plus some tweaks would make it. Regarding the BLE stack I am about to bring in Nordic's (Thisisant) S340 Softdevice bin blob, easy to integrate at Solo end? At Nordic end suppose that shall be via some placeholders already proposed by Nordic.
Regarding security and such don't mind, this is for evaluation - so if we even can get rid of BT pairing that's perfectly OK.

Ok I get it, let me study your answer during that weekend and discuss with my team. Good news you don't say no but rather are talking options, like that. Will come back quickly. Will check to hop into your link, thank you, cool! Cheers Chris

coelner, 2021-01-30:
FYI: https://github.com/google/OpenSK/ and related to that: https://github.com/google/OpenSK/issues/266

and furthermore: https://crates.io/crates/rubble-nrf52