solokeys/solo1 #502: Cannot register Solo Tap USB-C with Microsoft Account

matpk1990, 2021-01-23:
### Tested platforms/browsers:

- Fedora 33 (Vivaldi 3.5, Firefox 84)
- MacOS 11.1 (Microsoft Edge 87)

### Problem

I'm trying to register my Solo Tap USB-C keys with Microsoft Account (on https://account.microsoft.com) and i get an error message:
>We encountered an issue setting up your security key.
>We couldn't verify your identity or you are using private mode. Please ensure you are not in a private browsing window and please try again.`

I don't use private mode. I also tried to register keys using clean browser profiles. I found that error on Vivaldi stderr:

`[722004:722004:0123/142359.331330:ERROR:device_event_log_impl.cc(211)] [14:23:59.331] FIDO: authenticator_data.cc:72 CBOR decoding of authenticator data extensions failed (Map keys must be strictly monotonically increasing based on byte length and then by byte-wise lexical order.)`

timcappalli, 2021-01-23:
Known issue: https://github.com/solokeys/solo/issues/499

matpk1990, 2021-01-23:
Ok, thanks for quick response and sorry for duplicate issue.

timcappalli, 2021-01-23:
No worries! Glad more people are reporting it. I will add the error text from Microsoft Account to the other issue so it comes up in search.

conorpp, 2021-01-30:
Can you please try with newly released firmware 4.1.0?

matpk1990, 2021-01-30:
> Can you please try with newly released firmware 4.1.0?

I can register key after firmware update! Thanks for the fix.