solokeys/solo1 #503: Solokey Tap usb-c not detected

sarofim, 2021-01-24:
My solokey abruptly stopped working, both nfc & usb authentication no longer work.  I have the non-dev version, v2.1 hardware according to the pcb.

I've tested the key with the following devices:
- iPhone XR (via NFC)
- MacBook Pro 13" (via USB-C)

The green led that used to appear after a button press is no longer illuminated. I've inspected the PCB and can't see any obvious damage. 

The key is not detected by the `solo-python` library via `solo ls`.

It seems like the key just died completely. Any advice on how to debug this further ?

conorpp, 2021-01-25:
You can try to hold down the button for ~2s while plugging it in to see if it will enter bootloader mode, and then apply a firmware update.  If that doesn't work, I'm not sure how best to diagnose.  Can contact SoloKeys (hello@solokeys.com) for replacement.

sarofim, 2021-01-25:
Thanks for the quick response! I was able to enter bootloader mode and update the firmware to v4.0.0. 
USB-C authentication now works, but no luck with NFC sadly.

conorpp, 2021-01-30:
Was NFC working before?  You have to put it right above the "sweet spot" where the phone reads NFC.  On iPhones this is right on the bezel near the top/back of the phone.

sarofim, 2021-01-30:
NFC was working before. I have it near the sweet spot and have tried it when its powered or not.
The key is no longer via USB-C now so I've contacted SoloKeys for a replacement.

closed 2021-01-30 by sarofim