solokeys/solo1 #510: 4.1.0 not applicable on somu

wommel, 2021-01-30:
Trying to update to 4.1.0 results in error and leaves the somu unusable and blinking red.
```
$ solo key update
Not using FIDO2 interface.
Wrote temporary copy of firmware-4.1.0.json to /tmp/tmpgh1o51gm.json
sha256sums coincide: 039f963ed9c34a6fc5240b4597655da0cacdea501746f93bc6b3e9f0d69bef98
using signature version >2.5.3
erasing firmware...
updated firmware 100%
time: 9.37 s
bootloader is verifying signature...
...error!

Your key did not accept the firmware's signature! Possible reasons:
  * Tried to flash "hacker" firmware on custom hacker key with verifying bootloader

Currently, your key does not work. Please run update again with correct parameters
```

Although i was able to get the old version working again running `solo program bootloader firmware-4.0.0.json` it'd be nice to have the new version working.

kilaketia, 2021-01-30:
Same but with my Solo v1 (not hacker)
```
Not using FIDO2 interface.
Wrote temporary copy of firmware-4.1.0.json to /tmp/tmply4dl26q.json
sha256sums coincide: 039f963ed9c34a6fc5240b4597655da0cacdea501746f93bc6b3e9f0d69bef98
using signature version >2.5.3
erasing firmware...
updated firmware 100%             
time: 9.29 s
bootloader is verifying signature...
...error!

Your key did not accept the firmware's signature! Possible reasons:
  * Tried to flash "hacker" firmware on custom hacker key with verifying bootloader

Currently, your key does not work. Please run update again with correct parameters
```

conorpp, 2021-01-30:
Thanks for the quick report!  I've found the issue is the firmware version not getting inserted in the correct position in the binary, looking for a fix now.

conorpp, 2021-01-30:
It should be resolved now.

kilaketia, 2021-01-30:
Indeed, the update worked.

Thanks !

wommel, 2021-01-30:
thanks for the quick fix :)

closed 2021-01-30 by wommel