solokeys/solo1 #512: Ubuntu 20.04 - SoloKey not recognized after "solo key update"

rbortolameotti, 2021-01-30:
Hi everyone,

I have finally decided to upgrade the firmware of my key. After the (seemingly) successful update, the key is not recognized anymore by Ubuntu. Before the upgrade it was all working fine.

`solo key update
Not using FIDO2 interface.
Wrote temporary copy of firmware-4.1.0.json to /tmp/tmpex0f35jy.json
sha256sums coincide: 0b4a198bde48a4c0bbe6931ddf5fc24aa22aad5cd3d45391ef542d6c5ec7e724
using signature version <=2.5.3
erasing firmware...
updated firmware 100%             
time: 9.30 s
bootloader is verifying signature...
...pass!

Congratulations, your key was updated to the latest firmware version: 4.1.0
`

After the upgrade, they key does not blink or anything. 

`
$ solo ls
:: Solos
`

It is only recognized if put the key in Bootstrap mode, which still gives me hope about fixing the issue.

rbortolameotti, 2021-01-30:
Okay, after trying to install without the classic "update" command, but with program bootloader, it seems to be recognized again.

solo program bootloader ~/Downloads/firmware-4.1.0.json 
Not using FIDO2 interface.
using signature version <=2.5.3
erasing firmware...
updated firmware 100%             
time: 9.30 s

~$ solo ls
:: Solos
xxx: SoloKeys Solo 4.1.0

Tobi-K, 2021-02-01:
I have the exact same problem under Manjaro Linux with a "Solo Tap USB-A":
running `solo key update` looks like this:

> [tobi@bla]$ solo key update
> Not using FIDO2 interface.
> Wrote temporary copy of firmware-4.1.0.json to /tmp/tmpcyar3w_4.json
> sha256sums coincide: 0b4a198bde48a4c0bbe6931ddf5fc24aa22aad5cd3d45391ef542d6c5ec7e724
> using signature version <=2.5.3
> erasing firmware...
> updated firmware 100%             
> time: 9.31 s
> bootloader is verifying signature...
> ...pass!
> 
> Congratulations, your key was updated to the latest firmware version: 4.1.0

After that, there is no reaction at all when pluging the solo in, no LED, not shown with lusub, nothing in dmesg.
Bootloader mode (holding button for 2 seconds" works just fine, I can repeat the update process again and again.

The solution provided by @rbortolameotti unfortunately does not work for me, the results are the same.
I also tried the procedure under Windows, again with the same results.

Any help is greatly appreciated, thanks very much in advance.

conorpp, 2021-02-01:
@Tobi-K it looks like your key may be an old one that can be updated to older firmwares.  Can you try updating to [4.0.0](https://github.com/solokeys/solo/releases/download/4.0.0/firmware-4.0.0.json)?

Tobi-K, 2021-02-01:
It is indeed an older stick (shame on me for not using it earlier).
Updating to 4.0.0 first, then going to 4.1.0 did the trick and worked without problems.

Thank you very much for your fast help @conorpp .

rbortolameotti, 2021-02-01:
Indeed. I forgot to mention it myself. I also had an old key, and installing first 4.0.0 solved the problem.