solokeys/solo1 #514: Ubuntu 20.04 LTS : Solo key FIDO HID device timeout

gilbert-fernandes, 2021-01-30:
Already got a Yubikey. Bought a SoloKey to have a second key from another brand to protect my accounts
(I also did join the Solo2 kickstarter !)

When I plug my solo key, the light is green.
But I have not been able to have it work.

Checking the syslog I see this :

Jan 30 22:59:37 orion kernel: [62215.436945] usb 3-4: new full-speed USB device number 20 using xhci_hcd
Jan 30 22:59:37 orion kernel: [62215.586314] usb 3-4: New USB device found, idVendor=0483, idProduct=a2ca, bcdDevice= 1.00
Jan 30 22:59:37 orion kernel: [62215.586318] usb 3-4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Jan 30 22:59:37 orion kernel: [62215.586320] usb 3-4: Product: Solo 3.1.2
Jan 30 22:59:37 orion kernel: [62215.586322] usb 3-4: Manufacturer: SoloKeys
Jan 30 22:59:37 orion kernel: [62215.586323] usb 3-4: SerialNumber: 209A3793324B
Jan 30 22:59:37 orion kernel: [62215.589368] hid-generic 0003:0483:A2CA.0023: hiddev0,hidraw2: USB HID v1.11 Device [SoloKeys Solo 3.1.2] on usb-0000:00:14.0-4/input0
Jan 30 22:59:37 orion mtp-probe: checking bus 3, device 20: "/sys/devices/pci0000:00/0000:00:14.0/usb3/3-4"
Jan 30 22:59:37 orion mtp-probe: bus: 3, device: 20 was not an MTP device
Jan 30 22:59:38 orion fido_id[57054]: 3-4:1.0: Failed to open report descriptor at '/sys/devices/pci0000:00/0000:00:14.0/usb3/3-4/3-4:1.0/report_descriptor': No such file or directory
Jan 30 22:59:38 orion kernel: [62216.161821] hid-generic 0003:0483:A2CA.0024: hiddev0,hidraw2: USB HID v1.11 Device [SoloKeys Solo 3.1.2] on usb-0000:00:14.0-4/input0
Jan 30 22:59:38 orion mtp-probe: checking bus 3, device 20: "/sys/devices/pci0000:00/0000:00:14.0/usb3/3-4"
Jan 30 22:59:38 orion mtp-probe: bus: 3, device: 20 was not an MTP device

Everything seems OK but the "Failed to open descriptor" doesn't seem good.

After a few seconds, this appears :

Jan 30 22:59:43 orion gnome-shell[55743]: [55743:55743:0130/225943.958300:ERROR:device_event_log_impl.cc(208)] [22:59:43.958] FIDO: fido_hid_device.cc:518 FIDO HID device timeout for hid:d13efdd2-62aa-4cd7-ae23-a9baa0af3590
Jan 30 22:59:43 orion gnome-shell[55743]: [55743:55743:0130/225943.958346:ERROR:device_event_log_impl.cc(208)] [22:59:43.958] FIDO: ctap2_device_operation.h:126 -> (error reading)
Jan 30 22:59:43 orion gnome-shell[55743]: [55743:55743:0130/225943.958375:ERROR:device_event_log_impl.cc(208)] [22:59:43.958] FIDO: make_credential_request_handler.cc:524 Ignoring status 127 from hid:d13efdd2-62aa-4cd7-ae23-a9baa0af3590

I am getting timeouts from the device 

I don't know if there's something to do or if my key is defective.
But it's brand new and has never been used

gilbert-fernandes, 2021-01-30:
I have also tried this :

git clone the solo repo
cd udev
make setup

I did copy something but nothing has changed. Inserting the key still produces the same output in the syslog.
And the key cannot be used at 2FA from Chrome (where my other key does work).

gilbert-fernandes, 2021-01-30:
Ubuntu says the file is missing from there :

gf@orion:/sys/devices/pci0000:00/0000:00:14.0/usb3/3-4/3-4:1.0$ ls -l
total 0
drwxr-xr-x 4 root root    0 Jan 30 23:17 0003:0483:A2CA.0028
-rw-r--r-- 1 root root 4096 Jan 30 23:18 authorized
-r--r--r-- 1 root root 4096 Jan 30 23:18 bAlternateSetting
-r--r--r-- 1 root root 4096 Jan 30 23:17 bInterfaceClass
-r--r--r-- 1 root root 4096 Jan 30 23:17 bInterfaceNumber
-r--r--r-- 1 root root 4096 Jan 30 23:18 bInterfaceProtocol
-r--r--r-- 1 root root 4096 Jan 30 23:17 bInterfaceSubClass
-r--r--r-- 1 root root 4096 Jan 30 23:17 bNumEndpoints
lrwxrwxrwx 1 root root    0 Jan 30 23:17 driver -> ../../../../../../bus/usb/drivers/usbhid
drwxr-xr-x 3 root root    0 Jan 30 23:18 ep_01
drwxr-xr-x 3 root root    0 Jan 30 23:18 ep_81
lrwxrwxrwx 1 root root    0 Jan 30 23:18 firmware_node -> ../../../../../LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/device:16/device:17/device:1b
-r--r--r-- 1 root root 4096 Jan 30 23:18 interface
-r--r--r-- 1 root root 4096 Jan 30 23:18 modalias
drwxr-xr-x 2 root root    0 Jan 30 23:18 power
lrwxrwxrwx 1 root root    0 Jan 30 23:17 subsystem -> ../../../../../../bus/usb
-r--r--r-- 1 root root 4096 Jan 30 23:18 supports_autosuspend
-rw-r--r-- 1 root root 4096 Jan 30 23:17 uevent
drwxr-xr-x 3 root root    0 Jan 30 23:17 usbmisc

Seems to be looking for "report_descriptor" which does not exist.

gilbert-fernandes, 2021-01-30:
Contents of my 70-solokeys-access.rules in  /etc/udev/rules.d

# Notify ModemManager this device should be ignored
ACTION!="add|change|move", GOTO="mm_usb_device_blacklist_end"
SUBSYSTEM!="usb", GOTO="mm_usb_device_blacklist_end"
ENV{DEVTYPE}!="usb_device",  GOTO="mm_usb_device_blacklist_end"

ATTRS{idVendor}=="0483", ATTRS{idProduct}=="a2ca", ENV{ID_MM_DEVICE_IGNORE}="1"

LABEL="mm_usb_device_blacklist_end"

# Solo bootloader + firmware access
SUBSYSTEM=="hidraw", ATTRS{idVendor}=="0483", ATTRS{idProduct}=="a2ca", TAG+="uaccess"
SUBSYSTEM=="tty", ATTRS{idVendor}=="0483", ATTRS{idProduct}=="a2ca", TAG+="uaccess"

# ST DFU access
SUBSYSTEM=="usb", ATTRS{idVendor}=="0483", ATTRS{idProduct}=="df11", TAG+="uaccess"

# U2F Zero
SUBSYSTEM=="hidraw", ATTRS{idVendor}=="10c4", ATTRS{idProduct}=="8acf", TAG+="uaccess"

gilbert-fernandes, 2021-01-30:
Trying to use the key from Chrome, I start the request
Seems the request goes from Chrome to the device, which fails to answer :

```
Jan 30 23:35:57 orion gnome-shell[55743]: [55743:55743:0130/233557.346264:ERROR:device_event_log_impl.cc(208)] [23:35:57.346] FIDO: fido_hid_device.cc:518 FIDO HID device timeout for hid:264c0de5-a228-402e-bb1b-6c69b1413973
Jan 30 23:35:57 orion gnome-shell[55743]: [55743:55743:0130/233557.346307:ERROR:device_event_log_impl.cc(208)] [23:35:57.346] FIDO: ctap2_device_operation.h:126 -> (error reading)
Jan 30 23:35:57 orion gnome-shell[55743]: [55743:55743:0130/233557.346347:ERROR:device_event_log_impl.cc(208)] [23:35:57.346] FIDO: make_credential_request_handler.cc:524 Ignoring status 127 from hid:264c0de5-a228-402e-bb1b-6c69b1413973
```

Then Chrome also reports a timeout.

conorpp, 2021-01-30:
Can you try updating the firmware?  Install solo-python with `pip3 install solo-python`

Hold the button down, and plug it in while holding for ~2s.  It will change into bootloader mode with flashing orange light.

Then run "solo key update".

gilbert-fernandes, 2021-01-30:
Can't find the solo command.
Installed python3-pip then solo-python
Key is blinking orange
But "solo" is not a recognized command

gilbert-fernandes, 2021-01-30:
Seems it installed but "solo" is not available as a command :

gf@orion:/sys/devices/pci0000:00/0000:00:14.0/usb3/3-4/3-4:1.0$ pip3 install solo-python
Requirement already satisfied: solo-python in /home/gf/.local/lib/python3.8/site-packages (0.0.27)
Requirement already satisfied: intelhex in /home/gf/.local/lib/python3.8/site-packages (from solo-python) (2.3.0)
Requirement already satisfied: click>=7.0 in /usr/lib/python3/dist-packages (from solo-python) (7.0)
Requirement already satisfied: requests in /usr/lib/python3/dist-packages (from solo-python) (2.22.0)
Requirement already satisfied: pyserial in /home/gf/.local/lib/python3.8/site-packages (from solo-python) (3.5)
Requirement already satisfied: pyusb in /home/gf/.local/lib/python3.8/site-packages (from solo-python) (1.1.1)
Requirement already satisfied: cryptography in /usr/lib/python3/dist-packages (from solo-python) (2.8)
Requirement already satisfied: fido2~=0.8.1 in /home/gf/.local/lib/python3.8/site-packages (from solo-python) (0.8.1)
Requirement already satisfied: ecdsa in /home/gf/.local/lib/python3.8/site-packages (from solo-python) (0.16.1)
Requirement already satisfied: six in /usr/lib/python3/dist-packages (from fido2~=0.8.1->solo-python) (1.14.0)

conorpp, 2021-01-30:
I checked my ubuntu, and it installed to `~/.local/bin/solo`.  You could also install with root `sudo pip3 install solo-python` to add to global path I believe.

gilbert-fernandes, 2021-01-30:
No. I will not use root with Python.

I have been able to upgrade the key doing this :

1. sudo apt install python3-venv
2. python3 -m venv venv
3. source venv/bin/activate
4. echo solo-python > requirements.txt

After doing all of this, the "solo" command is now recognized and working

Upgrade :

(venv) gf@orion:~/solo-test$ solo key update
Not using FIDO2 interface.
Wrote temporary copy of firmware-4.1.0.json to /tmp/tmpyaom_j3c.json
sha256sums coincide: 0b4a198bde48a4c0bbe6931ddf5fc24aa22aad5cd3d45391ef542d6c5ec7e724
using signature version >2.5.3
erasing firmware...
updated firmware 100%             
time: 9.29 s
bootloader is verifying signature...
...pass!

Congratulations, your key was updated to the latest firmware version: 4.1.0

If I unplug/plug the key :

```
Jan 31 00:40:08 orion kernel: [68246.079229] usb 3-4: new full-speed USB device number 31 using xhci_hcd
Jan 31 00:40:08 orion kernel: [68246.232581] usb 3-4: New USB device found, idVendor=0483, idProduct=a2ca, bcdDevice= 1.00
Jan 31 00:40:08 orion kernel: [68246.232582] usb 3-4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Jan 31 00:40:08 orion kernel: [68246.232583] usb 3-4: Product: Solo 4.1.0
Jan 31 00:40:08 orion kernel: [68246.232584] usb 3-4: Manufacturer: SoloKeys
Jan 31 00:40:08 orion kernel: [68246.232585] usb 3-4: SerialNumber: 209A3793324B
Jan 31 00:40:08 orion kernel: [68246.234150] hid-generic 0003:0483:A2CA.0037: hiddev0,hidraw1: USB HID v1.11 Device [SoloKeys Solo 4.1.0] on usb-0000:00:14.0-4/input0
Jan 31 00:40:08 orion mtp-probe: checking bus 3, device 31: "/sys/devices/pci0000:00/0000:00:14.0/usb3/3-4"
Jan 31 00:40:08 orion mtp-probe: bus: 3, device: 31 was not an MTP device
Jan 31 00:40:09 orion fido_id[63514]: 3-4:1.0: Failed to open report descriptor at '/sys/devices/pci0000:00/0000:00:14.0/usb3/3-4/3-4:1.0/report_descriptor': No such file or directory
Jan 31 00:40:09 orion mtp-probe: checking bus 3, device 31: "/sys/devices/pci0000:00/0000:00:14.0/usb3/3-4"
Jan 31 00:40:09 orion mtp-probe: bus: 3, device: 31 was not an MTP device
Jan 31 00:40:09 orion kernel: [68246.803674] hid-generic 0003:0483:A2CA.0038: hiddev0,hidraw1: USB HID v1.11 Device [SoloKeys Solo 4.1.0] on usb-0000:00:14.0-4/input0
```
Adding the key using Chrome to my Github allowed keys :

Jan 31 00:41:47 orion gnome-shell[55743]: [55743:55743:0131/004147.027922:ERROR:device_event_log_impl.cc(208)] [00:41:47.027] FIDO: fido_hid_device.cc:518 FIDO HID device timeout for hid:f1836529-cdc7-4b14-afb5-c3eb192f1413
Jan 31 00:41:47 orion gnome-shell[55743]: [55743:55743:0131/004147.027993:ERROR:device_event_log_impl.cc(208)] [00:41:47.027] FIDO: ctap2_device_operation.h:126 -> (error reading)
Jan 31 00:41:47 orion gnome-shell[55743]: [55743:55743:0131/004147.028050:ERROR:device_event_log_impl.cc(208)] [00:41:47.028] FIDO: make_credential_request_handler.cc:524 Ignoring status 127 from hid:f1836529-cdc7-4b14-afb5-c3eb192f1413

Even with the latest fitmware and udev configured.

I'm going to reboot. Maybe the udev rules reload did not work as expected.

gilbert-fernandes, 2021-01-30:
After I reboot, plugging the key :

```
Jan 31 00:46:41 orion kernel: [  120.523335] usb 3-4: new full-speed USB device number 3 using xhci_hcd
Jan 31 00:46:41 orion kernel: [  120.673141] usb 3-4: New USB device found, idVendor=0483, idProduct=a2ca, bcdDevice= 1.00
Jan 31 00:46:41 orion kernel: [  120.673144] usb 3-4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Jan 31 00:46:41 orion kernel: [  120.673146] usb 3-4: Product: Solo 4.1.0
Jan 31 00:46:41 orion kernel: [  120.673147] usb 3-4: Manufacturer: SoloKeys
Jan 31 00:46:41 orion kernel: [  120.673148] usb 3-4: SerialNumber: 209A3793324B
Jan 31 00:46:41 orion mtp-probe: checking bus 3, device 3: "/sys/devices/pci0000:00/0000:00:14.0/usb3/3-4"
Jan 31 00:46:41 orion mtp-probe: bus: 3, device: 3 was not an MTP device
Jan 31 00:46:41 orion kernel: [  121.237829] usbcore: registered new interface driver usbhid
Jan 31 00:46:41 orion kernel: [  121.237830] usbhid: USB HID core driver
Jan 31 00:46:41 orion mtp-probe: checking bus 3, device 3: "/sys/devices/pci0000:00/0000:00:14.0/usb3/3-4"
Jan 31 00:46:41 orion mtp-probe: bus: 3, device: 3 was not an MTP device
Jan 31 00:46:41 orion kernel: [  121.250379] hid-generic 0003:0483:A2CA.0003: hiddev0,hidraw1: USB HID v1.11 Device [SoloKeys Solo 4.1.0] on usb-0000:00:14.0-4/input0
```

Trying to add the key to Github : Chrome tries to send the request to the key.
Timeouts, and fails.

```
Jan 31 00:50:13 orion gnome-shell[3878]: [3878:3878:0131/005013.340004:ERROR:device_event_log_impl.cc(208)] [00:50:13.339] FIDO: fido_hid_device.cc:518 FIDO HID device timeout for hid:44d1d833-ed0b-400f-a45b-f5a3195ce552
Jan 31 00:50:13 orion gnome-shell[3878]: [3878:3878:0131/005013.340056:ERROR:device_event_log_impl.cc(208)] [00:50:13.340] FIDO: ctap2_device_operation.h:126 -> (error reading)
Jan 31 00:50:13 orion gnome-shell[3878]: [3878:3878:0131/005013.340085:ERROR:device_event_log_impl.cc(208)] [00:50:13.340] FIDO: make_credential_request_handler.cc:524 Ignoring status 127 from hid:44d1d833-ed0b-400f-a45b-f5a3195ce552
```

No change.

conorpp, 2021-01-31:
I just tried on Ubuntu 20.04, Chrome, Github, firmware 4.1.0 and it worked fine.  

I don't think there is a USB issue with your device since it updated fine, and permissions seem to be okay.

Here is my dmesg:

```
[ 4208.641524] usb 1-10.4.4: new full-speed USB device number 19 using xhci_hcd
[ 4208.743277] usb 1-10.4.4: New USB device found, idVendor=0483, idProduct=a2ca, bcdDevice= 1.00
[ 4208.743283] usb 1-10.4.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 4208.743287] usb 1-10.4.4: Product: Solo 4.1.0
[ 4208.743290] usb 1-10.4.4: Manufacturer: SoloKeys
[ 4208.743293] usb 1-10.4.4: SerialNumber: 206532B33141
[ 4208.750557] hid-generic 0003:0483:A2CA.0012: hiddev4,hidraw9: USB HID v1.11 Device [SoloKeys Solo 4.1.0] on usb-0000:00:14.0-10.4.4/input0
[ 4208.768041] hid-generic 0003:0483:A2CA.0013: hiddev4,hidraw9: USB HID v1.11 Device [SoloKeys Solo 4.1.0] on usb-0000:00:14.0-10.4.4/input0

```

The main difference I see is your logs have mtp-probe popping in there.

```
Jan 31 00:40:08 orion mtp-probe: checking bus 3, device 31: "/sys/devices/pci0000:00/0000:00:14.0/usb3/3-4"
Jan 31 00:40:08 orion mtp-probe: bus: 3, device: 31 was not an MTP device
Jan 31 00:40:09 orion fido_id[63514]: 3-4:1.0: Failed to open report descriptor at '/sys/devices/pci0000:00/0000:00:14.0/usb3/3-4/3-4:1.0/report_descriptor': No such file or directory
```

Can you try killing this mtp-probe service and trying again?

gilbert-fernandes, 2021-02-01:
I have not yet been able to disable the mtp-probe.
Spent several hours searching with Google and trying dozens of dozens and totally differents things.
Nothing works.

No idea how to disable the mtp-probe if this is what makes the solokey not answer and end-up in timeout to chrome.

gilbert-fernandes, 2021-02-01:
I am not sure the mtp-probe is related.
The Yubikey I also see the message :+1: 

```
Feb  1 22:01:19 orion kernel: [37051.235941] hid-generic 0003:1050:0407.0013: input,hidraw1: USB HID v1.10 Keyboard [Yubico YubiKey OTP+FIDO+CCID] on usb-0000:00:14.0-8/input0
Feb  1 22:01:19 orion kernel: [37051.236609] hid-generic 0003:1050:0407.0014: hiddev0,hidraw2: USB HID v1.10 Device [Yubico YubiKey OTP+FIDO+CCID] on usb-0000:00:14.0-8/input1
Feb  1 22:01:19 orion mtp-probe: checking bus 3, device 12: "/sys/devices/pci0000:00/0000:00:14.0/usb3/3-8"
Feb  1 22:01:19 orion mtp-probe: bus: 3, device: 12 was not an MTP device
```

And it works fine.

Seems to me the mtp sees the device is not MTP for both Yubikey and Solokey
Thus, doesn't try to MTP it for anything.

I see Chrome call to the SoloKey and it ends up in timeout
So It seems to me the channel between chrome and SoloKey is working and data is being exchanged.

Iff MTP says in the log "not a mtp device" then MTP is not interfering I suppose ?

gilbert-fernandes, 2021-02-01:
I see this when I plug the solokey on Ubuntu 20.04 LTS + up to date firmware solokey :

```
Feb  1 22:17:54 orion kernel: [38046.221724] usb 3-8: new full-speed USB device number 13 using xhci_hcd
Feb  1 22:17:54 orion kernel: [38046.375090] usb 3-8: New USB device found, idVendor=0483, idProduct=a2ca, bcdDevice= 1.00
Feb  1 22:17:54 orion kernel: [38046.375092] usb 3-8: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Feb  1 22:17:54 orion kernel: [38046.375093] usb 3-8: Product: Solo 4.1.0
Feb  1 22:17:54 orion kernel: [38046.375093] usb 3-8: Manufacturer: SoloKeys
Feb  1 22:17:54 orion kernel: [38046.375094] usb 3-8: SerialNumber: 209A3793324B
Feb  1 22:17:54 orion kernel: [38046.376934] hid-generic 0003:0483:A2CA.0015: hiddev0,hidraw1: USB HID v1.11 Device [SoloKeys Solo 4.1.0] on usb-0000:00:14.0-8/input0
Feb  1 22:17:54 orion mtp-probe: checking bus 3, device 13: "/sys/devices/pci0000:00/0000:00:14.0/usb3/3-8"
Feb  1 22:17:54 orion mtp-probe: bus: 3, device: 13 was not an MTP device
Feb  1 22:17:54 orion fido_id[37121]: 3-8:1.0: Failed to open report descriptor at '/sys/devices/pci0000:00/0000:00:14.0/usb3/3-8/3-8:1.0/report_descriptor': No such file or directory
Feb  1 22:17:54 orion kernel: [38046.927703] hid-generic 0003:0483:A2CA.0016: hiddev0,hidraw1: USB HID v1.11 Device [SoloKeys Solo 4.1.0] on usb-0000:00:14.0-8/input0
Feb  1 22:17:54 orion mtp-probe: checking bus 3, device 13: "/sys/devices/pci0000:00/0000:00:14.0/usb3/3-8"
Feb  1 22:17:54 orion mtp-probe: bus: 3, device: 13 was not an MTP device
```

How come your system does not show the not found file part too ??

Linux orion 5.8.0-41-generic #46~20.04.1-Ubuntu SMP Mon Jan 18 17:52:23 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux

conorpp, 2021-02-01:
Hmm let's try another approach.  With the solo tool, can you try running the following commands?

```
solo key reset
# select yes, then press button
```

```
solo key make-credential
# No pin, then press button
```

gilbert-fernandes, 2021-02-01:
I see something different when trying to use Firefox instead of Chrome.
I ask Github to add the Solokey as key. The LED blinks between RED and GREEN
I press the button.. and it remains now stuck on RED. On Firefox, it is still waiting for me to press the button.

Pressing the button does nothing. And LED remains RED all the time now.
Log shows nothing. No timeout. Firefox waits, key remains RED after pressing when it was RED + GREEN blinking

gilbert-fernandes, 2021-02-01:
So I get this.

I plug the key. It's showing GREEN led.
Chrome : I try to add the key, the LED turns RED. I press the button, Nothing happens. Then timeout.

Firefox : I try to add the key, led blinks GREEN + RED. I press the button : remains RED and nothing happens.
Firefox waits forever, no timeout in syslog.

conorpp, 2021-02-01:
Can you try resetting the key and test make-credential?

```
solo key reset
# select yes, then press button
```

```
solo key make-credential
# No pin, then press button
```

gilbert-fernandes, 2021-02-01:
Tryint to reset the key :

```
gf@orion:~/.local/bin$ ./solo key reset 
Warning: Your credentials will be lost!!! Do you wish to continue? [y/N]: Y
Press the button to confirm -- again, your credentials will be lost!!!
^C
Aborted!
```

Key is green when plugged. After answering "y" it rurns RED and stays RED
Pressing the button on the key does nothing.
LED stays RED.
Nothing appears in syslog.

Second command :

solo key make-credential

gf@orion:~/.local/bin$ solo key make-credential
PIN (leave empty for no PIN): 
Touch your authenticator to generate a credential...

Key led remains green.nothing happens even if I press button.
syslog shows nothing.

gilbert-fernandes, 2021-02-01:
I plug the key. Use command to reset. LED is green. As soon I type Y and Enter led is red.
Pressing the button on solokey does nothing. Removing the key does this :

```
gf@orion:~/.local/bin$ ./solo key reset 
Warning: Your credentials will be lost!!! Do you wish to continue? [y/N]: y
Press the button to confirm -- again, your credentials will be lost!!!
Traceback (most recent call last):
  File "./solo", line 8, in <module>
    sys.exit(solo_cli())
  File "/usr/lib/python3/dist-packages/click/core.py", line 764, in __call__
    return self.main(*args, **kwargs)
  File "/usr/lib/python3/dist-packages/click/core.py", line 717, in main
    rv = self.invoke(ctx)
  File "/usr/lib/python3/dist-packages/click/core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/lib/python3/dist-packages/click/core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/lib/python3/dist-packages/click/core.py", line 956, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/lib/python3/dist-packages/click/core.py", line 555, in invoke
    return callback(*args, **kwargs)
  File "/home/gf/.local/lib/python3.8/site-packages/solo/cli/key.py", line 302, in reset
    solo.client.find(serial).reset()
  File "/home/gf/.local/lib/python3.8/site-packages/solo/client.py", line 221, in reset
    self.ctap2.reset()
  File "/home/gf/.local/lib/python3.8/site-packages/fido2/ctap2.py", line 795, in reset
    self.send_cbor(CTAP2.CMD.RESET, event=event, on_keepalive=on_keepalive)
  File "/home/gf/.local/lib/python3.8/site-packages/fido2/ctap2.py", line 642, in send_cbor
    response = self.device.call(CTAPHID.CBOR, request, event, on_keepalive)
  File "/home/gf/.local/lib/python3.8/site-packages/fido2/hid.py", line 89, in call
    status, resp = self._dev.InternalRecv()
  File "/home/gf/.local/lib/python3.8/site-packages/fido2/_pyu2f/hidtransport.py", line 308, in InternalRecv
    first_read = self.InternalReadFrame()
  File "/home/gf/.local/lib/python3.8/site-packages/fido2/_pyu2f/hidtransport.py", line 300, in InternalReadFrame
    frame = self.hid_device.Read()
  File "/home/gf/.local/lib/python3.8/site-packages/fido2/_pyu2f/linux.py", line 226, in Read
    raw_in = os.read(self.dev, self.GetInReportDataLength())
OSError: [Errno 5] Input/output error
```

conorpp, 2021-02-01:
Ok I'm not sure what's going on.  The LED should change orange/red, but then a button press should clear it, returning back to green.  For some reason the button isn't working, or perhaps the flash on the device is corrupted.  If you send hello@solokeys.com an email with your shipping or order info, we'll gladly send you a replacement.

gilbert-fernandes, 2021-02-01:
If I try the credentiaL :

```
gf@orion:~/.local/bin$ solo key make-credential
PIN (leave empty for no PIN): 
Touch your authenticator to generate a credential...
Traceback (most recent call last):
  File "/home/gf/.local/bin/solo", line 8, in <module>
    sys.exit(solo_cli())
  File "/usr/lib/python3/dist-packages/click/core.py", line 764, in __call__
    return self.main(*args, **kwargs)
  File "/usr/lib/python3/dist-packages/click/core.py", line 717, in main
    rv = self.invoke(ctx)
  File "/usr/lib/python3/dist-packages/click/core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/lib/python3/dist-packages/click/core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/lib/python3/dist-packages/click/core.py", line 956, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/lib/python3/dist-packages/click/core.py", line 555, in invoke
    return callback(*args, **kwargs)
  File "/home/gf/.local/lib/python3.8/site-packages/solo/cli/key.py", line 151, in make_credential
    solo.hmac_secret.make_credential(
  File "/home/gf/.local/lib/python3.8/site-packages/solo/hmac_secret.py", line 58, in make_credential
    attestation_object, client_data = client.make_credential(options, pin=pin)
  File "/home/gf/.local/lib/python3.8/site-packages/fido2/client.py", line 395, in make_credential
    self._do_make_credential(
  File "/home/gf/.local/lib/python3.8/site-packages/fido2/client.py", line 459, in _ctap2_make_credential
    return self.ctap2.make_credential(
  File "/home/gf/.local/lib/python3.8/site-packages/fido2/ctap2.py", line 690, in make_credential
    return self.send_cbor(
  File "/home/gf/.local/lib/python3.8/site-packages/fido2/ctap2.py", line 642, in send_cbor
    response = self.device.call(CTAPHID.CBOR, request, event, on_keepalive)
  File "/home/gf/.local/lib/python3.8/site-packages/fido2/hid.py", line 89, in call
    status, resp = self._dev.InternalRecv()
  File "/home/gf/.local/lib/python3.8/site-packages/fido2/_pyu2f/hidtransport.py", line 308, in InternalRecv
    first_read = self.InternalReadFrame()
  File "/home/gf/.local/lib/python3.8/site-packages/fido2/_pyu2f/hidtransport.py", line 300, in InternalReadFrame
    frame = self.hid_device.Read()
  File "/home/gf/.local/lib/python3.8/site-packages/fido2/_pyu2f/linux.py", line 226, in Read
    raw_in = os.read(self.dev, self.GetInReportDataLength())
OSError: [Errno 5] Input/output error
```

Key remains green after pressing enter when it asks for a pin.
Pressing the button on key does nothing.

Error above appears if I physically remove it.

gilbert-fernandes, 2021-02-01:
Maybe my button on the key is defective. I dont know :(

Sending you an email. Bought mine on Amazon

conorpp, 2021-02-01:
Do you happen to have a second authenticator plugged in?  Either yubikey, or built-in authenticator on laptop (does fido2 work when no keys are plugged in?).

gilbert-fernandes, 2021-02-01:
I have two keys : one ubikey 5 and the solokey.
I only have two plugs on this machine (a Dell XPS 2020 9300). I got one left, and one right of machine
Both are Thunderbolt 3/USB-C holes.
Right has power USB-C so I use left plug

I also tried to remove the right power supply and use the right hole for the solokey

I never used both keys at the same time.

gilbert-fernandes, 2021-02-01:
I just tried. Yubikey on left plug.
Firefox and Chrome : I can log out of Github and then Login back and key blinks, touching it unlocks account.
So I think FIDO2 works on both browsers, and Ubuntu seems OK.

Seems my Solokey button is not working. Or the key just stays RED-Led when it was blinking orange/red and that only happens when using Firefox. When using Chrome, it becomes red, and pressing it does nothing.

gilbert-fernandes, 2021-02-01:
Took pics of my key but nothing seems wrong on it :

https://ibb.co/JQ1WqDV
https://ibb.co/28xg4NX

gilbert-fernandes, 2021-02-09:
Received the new key today.
Worked immediatly with NO problem at all :)

When plugged :

Feb  9 14:19:27 orion kernel: [  239.352958] usb 3-4: new full-speed USB device number 4 using xhci_hcd
Feb  9 14:19:28 orion kernel: [  239.503653] usb 3-4: New USB device found, idVendor=0483, idProduct=a2ca, bcdDevice= 1.00
Feb  9 14:19:28 orion kernel: [  239.503656] usb 3-4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Feb  9 14:19:28 orion kernel: [  239.503658] usb 3-4: Product: Solo 3.1.2
Feb  9 14:19:28 orion kernel: [  239.503659] usb 3-4: Manufacturer: SoloKeys
Feb  9 14:19:28 orion kernel: [  239.503661] usb 3-4: SerialNumber: XXXXXXXXXXXX
Feb  9 14:19:28 orion kernel: [  239.507447] hid-generic 0003:0483:A2CA.0005: hiddev0,hidraw1: USB HID v1.11 Device [SoloKeys Solo 3.1.2] on usb-0000:00:14.0-4/input0
Feb  9 14:19:28 orion mtp-probe: checking bus 3, device 4: "/sys/devices/pci0000:00/0000:00:14.0/usb3/3-4"
Feb  9 14:19:28 orion mtp-probe: bus: 3, device: 4 was not an MTP device
Feb  9 14:19:28 orion fido_id[5780]: 3-4:1.0: Failed to open report descriptor at '/sys/devices/pci0000:00/0000:00:14.0/usb3/3-4/3-4:1.0/report_descriptor': No such file or directory
Feb  9 14:19:28 orion kernel: [  240.081648] hid-generic 0003:0483:A2CA.0006: hiddev0,hidraw1: USB HID v1.11 Device [SoloKeys Solo 3.1.2] on usb-0000:00:14.0-4/input0
Feb  9 14:19:28 orion mtp-probe: checking bus 3, device 4: "/sys/devices/pci0000:00/0000:00:14.0/usb3/3-4"
Feb  9 14:19:28 orion mtp-probe: bus: 3, device: 4 was not an MTP device

This confirms that the presence of the "not a MTP device" is normal and does not, on Ubuntu, create a problem.
So the udev file that is given by the project works perfectly fine as is on Ubuntu 20.04 LTS
Went into Github, added the solo key
It turned RED, pressed the button and immediatly worked :D

SoloKey v1 works perfectly on Ubuntu 20.04 LTS
I only did need to git clone the udev repo and use it to create the required udev file
(which works without any alteration)

Woo Woo :D

gilbert-fernandes, 2021-02-10:
Issue fixed : problem with key. Sending key back to N******* in Switzerland so it can be checked.

closed 2021-02-10 by gilbert-fernandes