solokeys/solo1 #515: ssh-add -K not working on macOS

max-wittig, 2021-02-02:
Split up issue from https://github.com/solokeys/solo/issues/374, as there is too much other things going on there

--- 

I'm not sure, if this is the right place but my team and me can't get `ssh-add` and `ssh-add -K` to work with the Solokey.
`ssh-agent` is just reporting `Unable to add key ED25519-SK SHA256:XXXXXXXXXXX`. More information below.

We've tried with `resident` and `non-resident` keys.

We're using the openssh from homebrew on macOS

### OpenSSH version

```sh
max@computer /tmp % ssh-add -V
OpenSSH_8.4p1, OpenSSL 1.1.1i  8 Dec 2020
```

### Running the failed `ssh-add -K` command

```sh
max@computer /tmp % ssh-add -v -K
Enter PIN for authenticator: 
debug1: start_helper: starting /usr/local/Cellar/openssh/8.4p1_2/libexec/ssh-sk-helper 
debug1: sshsk_load_resident: provider "internal", have-pin
debug1: sk_probe: 1 device(s) detected
debug1: sk_probe: selecting sk by touch
debug1: ssh_sk_load_resident_keys: trying IOService:/AppleACPIPlatformExpert/PCI0@0/AppleACPIPCI/XHC1@14/XHC1@14000000/HS07@14200000/Solo 4.1.0@14200000/Solo 4.1.0@0/AppleUserUSBHostHIDDevice
debug1: read_rks: existing 1, remaining 49
debug1: read_rks: Device IOService:/AppleACPIPlatformExpert/PCI0@0/AppleACPIPCI/XHC1@14/XHC1@14000000/HS07@14200000/Solo 4.1.0@14200000/Solo 4.1.0@0/AppleUserUSBHostHIDDevice has resident keys for 1 RPs
debug1: read_rks: rp 0: name="openssh" id="ssh:" hashlen=32
debug1: read_rks: RP "ssh:" has 1 resident keys
debug1: read_rks: Device IOService:/AppleACPIPlatformExpert/PCI0@0/AppleACPIPCI/XHC1@14/XHC1@14000000/HS07@14200000/Solo 4.1.0@14200000/Solo 4.1.0@0/AppleUserUSBHostHIDDevice RP "ssh:" slot 0: type -8 flag
s 0x00 prot 0x01
debug1: process_load_resident: key 0 ED25519-SK ssh:
debug1: ssh-sk-helper: reply len 165
debug1: sshsk_load_resident: keys[0]: ED25519-SK ssh:
Unable to add key ED25519-SK SHA256:XXXXXXXXXXX
```

enrikb, 2021-02-02:
As already mentioned on #374, it works for me pretty well on Ubuntu 20.10. The verbose output looks pretty similar.
Only the last line differs, showing success.

Could it be that your _agent_ does not support the key type? Does macOS bring its own ssh agent?

max-wittig, 2021-02-02:
Ohhh man. Thanks for this hint. Woah, I've just not restarted the `ssh-agent`. Spend hours with this already.

For other that run into the same problem:

```sh
killall ssh-agent
eval $(ssh-agent)
ssh-add -K
```

Thank you very much! 👍 💚

closed 2021-02-02 by max-wittig

albert-a, 2021-02-20:
Thanks! It also helped me! Just reloaded ssh-agent.