solokeys/solo1 #518: Updating Solo Tap USB-A not working

panache67, 2021-02-07:
I try to update a Solo Tap USB-A running version 3.0 using the solo-python. 
After a success message, the Solo is no longer detected and putting it back into flashing modus states that the Solo is still running version 3.0. Below the logs: 

> (base) PS C:\Windows\system32> solo key version
Not using FIDO2 interface.
3.0.0 locked
(base) PS C:\Windows\system32> solo key update
Not using FIDO2 interface.
Wrote temporary copy of firmware-4.1.0.json to C:\Users\xxxxxxx\AppData\Local\Temp\tmpctcg3t_i.json
sha256sums coincide: 0b4a198bde48a4c0bbe6931ddf5fc24aa22aad5cd3d45391ef542d6c5ec7e724
using signature version >2.5.3
erasing firmware...
updated firmware 100%
time: 9.29 s
bootloader is verifying signature...
...pass!
Congratulations, your key was updated to the latest firmware version: 4.1.0

>> (base) PS C:\Windows\system32> solo key version
Not using FIDO2 interface.
3.0.0 locked

Any help is very much appreciated.

Trapacid, 2021-02-07:
Try manually programming it.

https://github.com/solokeys/solo/releases/download/4.1.1/firmware-4.1.1.json

`solo program bootloader firmware-4.1.1.json`

panache67, 2021-02-07:
Awesome that worked immediately and solved the issue. Thank you @Trapacid

closed 2021-02-07 by panache67