solokeys/solo1 #521: Ubuntu 20.04 LTS - 'ssh-keygen -v -t ed25519-sk -Ono-touch-required' : Key enrollment failed: invalid format (Solokeys)

steffen835, 2021-02-22:
I am using the latest Firmware 4.1.1:

steffen@steffen-laptop:~$ solo key reset
Warning: Your credentials will be lost!!! Do you wish to continue? [y/N]: y
Press the button to confirm -- again, your credentials will be lost!!!
....aaaand they're gone
steffen@steffen-laptop:~$

but I still can't add a key using ssh-keygen. I tried with and without a PIN:

without PIN:

steffen@steffen-laptop:~$ ssh-keygen -v -t ed25519-sk -Ono-touch-required -C steffen-solokey_notouch -f /home/steffen/.ssh/steffen-solokey_notouch
Generating public/private ed25519-sk key pair.
You may need to touch your authenticator to authorize key generation.
debug1: start_helper: starting /usr/lib/openssh/ssh-sk-helper 
debug1: sshsk_enroll: provider "internal", device "(null)", application "ssh:", userid "(null)", flags 0x00, challenge len 0
debug1: sshsk_enroll: using random challenge
debug1: ssh_sk_enroll: using device /dev/hidraw4
debug1: ssh_sk_enroll: fido_cred_verify: FIDO_ERR_INVALID_SIG
debug1: sshsk_enroll: provider "internal" returned failure -1
debug1: ssh-sk-helper: Enrollment failed: invalid format
debug1: ssh-sk-helper: reply len 8
debug1: client_converse: helper returned error -4
Key enrollment failed: invalid format
steffen@steffen-laptop:~$ 

with PIN:

steffen@steffen-laptop:~$ solo key set-pin
Please enter new pin: 
Please confirm new pin: 
Done. Please use new pin to verify key
steffen@steffen-laptop:~$ ssh-keygen -v -t ed25519-sk -Ono-touch-required -C steffen-solokey_notouch -f /home/steffen/.ssh/steffen-solokey_notouch
Generating public/private ed25519-sk key pair.
You may need to touch your authenticator to authorize key generation.
debug1: start_helper: starting /usr/lib/openssh/ssh-sk-helper 
debug1: sshsk_enroll: provider "internal", device "(null)", application "ssh:", userid "(null)", flags 0x00, challenge len 0
debug1: sshsk_enroll: using random challenge
debug1: ssh_sk_enroll: using device /dev/hidraw4
debug1: ssh_sk_enroll: fido_dev_make_cred: FIDO_ERR_PIN_REQUIRED
debug1: sshsk_enroll: provider "internal" returned failure -3
debug1: ssh-sk-helper: Enrollment failed: incorrect passphrase supplied to decrypt private key
debug1: ssh-sk-helper: reply len 8
debug1: client_converse: helper returned error -43
Enter PIN for authenticator: 
debug1: start_helper: starting /usr/lib/openssh/ssh-sk-helper 
debug1: sshsk_enroll: provider "internal", device "(null)", application "ssh:", userid "(null)", flags 0x00, challenge len 0 with-pin
debug1: sshsk_enroll: using random challenge
debug1: ssh_sk_enroll: using device /dev/hidraw4
debug1: ssh_sk_enroll: fido_cred_verify: FIDO_ERR_INVALID_SIG
debug1: sshsk_enroll: provider "internal" returned failure -1
debug1: ssh-sk-helper: Enrollment failed: invalid format
debug1: ssh-sk-helper: reply len 8
debug1: client_converse: helper returned error -4
Key enrollment failed: invalid format
steffen@steffen-laptop:~$ 

The stick nonetheless works on https://webauthn.io/ (latest Chrome).

My Ubuntu 20.04 LTS is up to date. I even tried in a freshly installed and up-to-date Ubunto 20.04 LTS VM - same result.

Adding an ssh key this way is working for my colleagues who also use Ubuntu 20.04 LTS and a Solokeys stick.

uli-heller, 2021-02-22:
Hey @steffen835 , I asked another team member to

- update to firmware 4.1.1
- create an ed25519-sk key

Worked without an issue. Actually, the hardware is from the same batch - I ordered 8 solokeys and gave one of them to you and another one to the other team member. Strange, isn't it?

steffen835, 2021-02-24:
Some additional observations:

Here ist the data that gets passed to load_der_x509_certificate when trying to verify the solokeys secure:

/usr/lib/python3/dist-packages/cryptography/hazmat/backends/openssl/backend.py:

    def load_der_x509_certificate(self, data):
        f = open('/home/steffen/x/data', 'wb')
        f.write(data)
        f.close()
        mem_bio = self._bytes_to_bio(data)
        x509 = self._lib.d2i_X509_bio(mem_bio.bio, self._ffi.NULL)
        if x509 == self._ffi.NULL:
            self._consume_errors()
            raise ValueError("Unable to load certificate")

        x509 = self._ffi.gc(x509, self._lib.X509_free)
        return _Certificate(self, x509)

==>

steffen@steffen-laptop:~/x$ hexdump -v -C data
00000000  30 82 02 e1 30 82 02 88  a0 03 02 01 02 02 01 01  |0...0...........|
00000010  30 0a 06 08 2a 86 48 ce  3d 04 03 02 30 81 80 31  |0...*.H.=...0..1|
00000020  0b 30 09 06 03 55 04 06  13 02 55 53 31 11 30 0f  |.0...U....US1.0.|
00000030  06 03 55 04 08 0c 08 4d  61 72 79 6c 61 6e 64 31  |..U....Maryland1|
00000040  12 30 10 06 03 55 04 0a  0c 09 53 6f 6c 6f 20 4b  |.0...U....Solo K|
00000050  65 79 73 31 10 30 0e 06  03 55 04 0b 0c 07 52 6f  |eys1.0...U....Ro|
00000060  6f 74 20 43 41 31 15 30  13 06 03 55 04 03 0c 0c  |ot CA1.0...U....|
00000070  73 6f 6c 6f 6b 65 79 73  2e 63 6f 6d 31 21 30 1f  |solokeys.com1!0.|
00000080  06 09 2a 86 48 86 f7 0d  ff ff ff ff ff ff ff ff  |..*.H...........|
00000090  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
000000a0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
000000b0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
000000c0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
000000d0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
000000e0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
000000f0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
00000100  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
00000110  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
00000120  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
00000130  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
00000140  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
00000150  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
00000160  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
00000170  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
00000180  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
00000190  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
000001a0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
000001b0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
000001c0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
000001d0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
000001e0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
000001f0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
00000200  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
00000210  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
00000220  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
00000230  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
00000240  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
00000250  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
00000260  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
00000270  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
00000280  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
00000290  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
000002a0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
000002b0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
000002c0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
000002d0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
000002e0  ff ff ff ff ff                                    |.....|
000002e5
steffen@steffen-laptop:~/x$

The solokeys secure key does work for https://webauthn.io/ but it does not work for https://demo.yubico.com/webauthn-technical/registration
When I try to use it on the yubico site I get this error:

Maybe the operation timed out or was aborted. Below is the error that was provided by the web browser:

Unable to load certificate

I also tried this with the latest Firefox on Windows 10 (20H2) with the same result.

steffen835, 2021-02-24:
The test  fails too:
```
echo credential challenge | openssl sha256 -binary | base64 > cred_param
echo relying party >> cred_param
echo user name >> cred_param
dd if=/dev/urandom bs=1 count=32 | base64 >> cred_param
fido2-cred -M -i cred_param /dev/hidraw4 | fido2-cred -V -d -o cred
Enter PIN for /dev/hidraw4: 
client data hash:
  61 2e ef 12 ca 69 87 b3 16 c0 61 8b 3a 24 21 a4 
  3c 7e 59 8c 8a 99 2f 3a 44 e1 85 3c 3f 7d 02 71 
relying party id: relying party
format: packed
authenticator data:
  58 ca ae b0 38 84 97 c8 c3 d3 75 c1 57 ee 72 06 
  98 ac 78 78 be 87 0a d8 f1 aa 99 37 2f ac 5d b4 
  5b 54 45 00 00 06 7d 88 76 63 1b d4 a0 42 7f 57 
  73 0e c7 1c 9e 02 79 00 46 7b 35 68 1c 44 25 2e 
  e2 01 f8 97 e3 76 46 af 1a 40 a2 26 67 c9 f8 c0 
  9a 2b ca 80 10 a0 07 ad c9 5b d0 ae b0 38 84 97 
  c8 c3 d3 75 c1 57 ee 72 06 98 ac 78 78 be 87 0a 
  d8 f1 aa 99 37 2f ac 5d b4 5b 54 7d 06 00 00 a5 
  01 02 03 26 20 01 21 58 20 2d 61 c2 77 c4 4a 1b 
  4f 69 60 bf 57 43 fd b6 64 ac 9c 5e 08 e7 c4 ad 
  c4 2b 0c a5 c8 6e 17 b4 45 22 58 20 1d 77 d9 e5 
  13 31 37 7a 32 57 46 04 3c b6 7d 85 55 df 07 92 
  8d d3 c5 39 84 e7 e3 54 7a 27 93 f8 
credential id:
  7b 35 68 1c 44 25 2e e2 01 f8 97 e3 76 46 af 1a 
  40 a2 26 67 c9 f8 c0 9a 2b ca 80 10 a0 07 ad c9 
  5b d0 ae b0 38 84 97 c8 c3 d3 75 c1 57 ee 72 06 
  98 ac 78 78 be 87 0a d8 f1 aa 99 37 2f ac 5d b4 
  5b 54 7d 06 00 00 
signature:
  30 45 02 20 09 ef 13 9f ad 7b fd 7f 3e 3e 2e 7a 
  ea cf 2e 75 25 b1 30 32 48 1d 6e eb eb 3a bb 97 
  72 b3 7c 32 02 21 00 c2 fc 47 cf 47 d6 c8 a1 93 
  2a c9 ee 6c db 1a e1 b6 b6 19 f1 00 f3 ae 4f cc 
  0e d8 87 ad f6 ae aa 
x509:
  30 82 02 e1 30 82 02 88 a0 03 02 01 02 02 01 01 
  30 0a 06 08 2a 86 48 ce 3d 04 03 02 30 81 80 31 
  0b 30 09 06 03 55 04 06 13 02 55 53 31 11 30 0f 
  06 03 55 04 08 0c 08 4d 61 72 79 6c 61 6e 64 31 
  12 30 10 06 03 55 04 0a 0c 09 53 6f 6c 6f 20 4b 
  65 79 73 31 10 30 0e 06 03 55 04 0b 0c 07 52 6f 
  6f 74 20 43 41 31 15 30 13 06 03 55 04 03 0c 0c 
  73 6f 6c 6f 6b 65 79 73 2e 63 6f 6d 31 21 30 1f 
  06 09 2a 86 48 86 f7 0d ff ff ff ff ff ff ff ff 
  ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
  ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
  ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
  ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
  ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
  ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
  ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
  ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
  ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
  ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
  ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
  ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
  ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
  ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
  ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
  ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
  ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
  ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
  ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
  ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
  ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
  ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
  ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
  ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
  ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
  ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
  ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
  ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
  ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
  ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
  ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
  ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
  ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
  ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
  ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
  ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
  ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
  ff ff ff ff ff 
cbor_decode_cred_authdata: buf=0x55557f286250, len=202
decode_attcred: buf=0x55557f286275, len=165
decode_attcred: attcred->id.len=70
fido_check_flags: flags=45
fido_check_flags: up=2, uv=0
verify_sig: x509 key
fido_cred_verify: verify_sig
fido2-cred: fido_cred_verify: FIDO_ERR_INVALID_SIG
```

enrikb, 2021-02-24:
I just did the same test using a Solo Hacker and it worked (expected).
What looks very strange in your dump are the many `ff`s in the x509 part.
I read that you also tried using Windows. But did you try another PC, as well? Maybe it's a hardware issue or whatever kind of USB hickup?

enrikb, 2021-02-24:
... or the flash was not written properly and the `ff`s actually come from the stored certificate?

steffen835, 2021-02-25:
I already reflashed the firmware with 'solo key update' but that did not fix the problem: 
```
steffen@steffen-laptop:~$ solo key update
Wrote temporary copy of firmware-4.1.1.json to /tmp/tmpfr871kms.json
sha256sums coincide: 45e7071107fa5acb494e4c5d4831344dd531b1a537a230a6157f4f71dba87e2c
Switching into bootloader mode...
Not using FIDO2 interface.
using signature version >2.5.3
erasing firmware...
updated firmware 100%             
time: 14.02 s
bootloader is verifying signature...
...pass!

Congratulations, your key was updated to the latest firmware version: 4.1.1
steffen@steffen-laptop:~$ 
```

It really looks like the attestation key got damaged.

According to the sourcecode the attestation data has its own page outside of the application region:
```
solo/targets/stm32l432/src/memory_layout.h:

#define PAGE_SIZE		2048
#define PAGES			128

// Pages 119-127 are data
// Location of counter page and it's backup page
// The flash is wear leveled and counter should be fault tolerant
...
// where attestation key is located
#define ATTESTATION_PAGE        (PAGES - 15)
#define ATTESTATION_PAGE_ADDR   (0x08000000 + ATTESTATION_PAGE*PAGE_SIZE)
...
#define ATTESTATION_CONFIGURED_TAG      0xaa551e79

struct flash_attestation_page{
  uint8_t attestation_key[32];
  // DWORD padded.
  uint64_t device_settings;
  uint64_t attestation_cert_size;
  uint8_t attestation_cert[2048 - 32 - 8 - 8];
} __attribute__((packed));

typedef struct flash_attestation_page flash_attestation_page;

static_assert(sizeof(flash_attestation_page) == 2048, "Data structure doesn't match flash size");
```
When plugging in the stick for the first time the attestation certificate and lock information gets copied from the application region to this separate attestation page and the ATTESTATION_CONFIGURED_TAG gets set preventing further copying:
```
    uint64_t device_settings = ((flash_attestation_page *)ATTESTATION_PAGE_ADDR)->device_settings;
    uint32_t configure_tag = (uint32_t)(device_settings >> 32);

    if (configure_tag != ATTESTATION_CONFIGURED_TAG)
    {
        printf1(TAG_RED,"Migrating certificate and lock information to data segment.\r\n");

        device_settings = ATTESTATION_CONFIGURED_TAG;
        device_settings <<= 32;
       ...
    }
```
   This page does not seem to get reflashed by a 'solo key reset'.
   ==> It does not seem possible to fix this issue for a solokeys secure key without a firmware fix/workaround for this
          particular problem.

steffen835, 2021-02-26:
Is the solokeys secure stick defective?
Should I rma it and ask for a replacement?

conorpp, 2021-02-28:
Hey @steffen835, thanks for reporting & sorry for delay.  It definitely seems like the attestation certificate is only partially programmed.  Please send a request to hello@solokeys.com for a replacement.

closed 2021-02-28 by conorpp