solokeys/solo1 #523: Question: AAGUID

FaganSC, 2021-02-26:
Is there a way to obtain the AAGUID of a Solo Tap? Trying to configure my Solo Tap with Azure AD, and they now have the ability to add organizational approved devices if you have the AAGUID .

[image]

aseigler, 2021-02-27:
See https://docs.solokeys.dev/metadata-statements/

FaganSC, 2021-02-27:
I'm thinking it might take a little bit for Microsoft to push it out because I'm still getting the following error message after putting the AAGUID in for Azure AD.

[image]

aseigler, 2021-02-27:
It definitely works if you turn off the key restrictions.

back2root, 2021-06-03:
@FaganSC checkout: https://github.com/solokeys/solo/tree/master/metadata
Solo Tap should have aaguid: 8976631b-d4a0-427f-5773-0ec71c9e0279

closed 2022-06-17 by FaganSC