solokeys/solo1 #525: Solokey hardware V2.1 - No LEDs, in Ubuntu 20.10 or Windows 10

SpindlyOxtail, 2021-02-28:
Solokey hardware V2.1 - No LEDs, in Ubuntu 20.10 or Windows 10

In Windows 10 it shows as 'Universal Serial bus devices' STM32 Bootloader

I have tried different USB drivers using 'Zadig' and also tried using 'solo-python' commands to update firmware but nothing works. It always says no solo key...

C:\Windows\system32>solo key update
No Solo key found!

I can send you what appears in Ubuntu if needed aswell.

Please help - Im gettting a refund if not as i have 4 and they all do same thing

Cheers

msiqx9, 2021-03-04:
Best to get a refund whilst you can.  I have a v2.1 Solo Tap and have terrible NFC problems.

conorpp, 2021-03-04:
Showing up as "STM32 Bootloader" indicates that they have not been programmed.  If you're inclined you can program them using the following.

```
wget https://github.com/solokeys/solo/releases/download/4.1.1/bundle-secure-non-solokeys-4.1.1.hex
solo program dfu bundle-secure-non-solokeys-4.1.1.hex
solo program aux leave-dfu
```

Please contact hello@solokeys.com with information on where you got the device and for refund.

closed 2021-03-04 by conorpp