solokeys/solo1 #530: solokey not accepted by windows 10 pro v.20H2

Bob-code-dot, 2021-03-12:
hello, my (new) security key doesn't work on my computer with windows 10 pro version 20H2. I tried to use this link "https://update.solokeys.com/", but I cannot because my computer don't "accept" this key, it's written "impossible tu read your secutity key. try again.".
Can you help me please?
[image]

msiqx9, 2021-03-13:
Try using the Windows [Solo Desktop ](https://github.com/solokeys/solo-desktop)updater.

Bob-code-dot, 2021-03-13:
I understand absolutely nothing about all of that. Do you have one solution or I refund it?

msiqx9, 2021-03-13:
Then refund it.

Bob-code-dot, 2021-03-13:
But I suppose you are just one person trying to help me, right? because if I am here it's to speak with the manufacturer... I asked him one solution on Amazon, 2 times, and the only one answer was "ask it on Github"...
Can you confirm me?

msiqx9, 2021-03-13:
I'm just another customer who bought the Solo Tap v2.1 on Amazon at the end of Jan 2021 and is struggling with NFC problems; I've yet to find a solution.  You are in the right place on Github...

aseigler, 2021-03-17:
The update site hasn't worked in a while. You should be able to follow the instructions at https://docs.solokeys.dev/programming/ with the release packages at https://github.com/solokeys/solo/releases to update.

dnorthup-ums, 2021-09-13:
The solution for this and #529 are identical, presuming the issue description is complete in each case.