solokeys/solo1 #536: How to the solo key?

markusqw1, 2021-03-19:
I’m new to solo key. How do I check for updates and update the solo key?

msiqx9, 2021-03-19:
I suggest installing [solo-desktop](https://github.com/solokeys/solo-desktop) to your W10 PC and use that to update.

silvernode, 2021-03-23:
I am wondering the same thing. I have had my key for over a year now and wonder if I have to flash any new firmware on the device.

msiqx9, 2021-03-28:
You could try using the [Solo Webupdate](https://update.solokeys.com/) or the [Solo Desktop](https://github.com/solokeys/solo-desktop).

ccoenen, 2022-02-16:
Solo **WebUpdate** currently does not work because "Site firmware out of date"
[image]

The Solo **Desktop** is marked as "WIP", the release is "First Release" from 2019. Hardly inspiring confidence.
[image]

The python package **solo-python** requires administrative privileges on windows
[image]

On Ubuntu linux, the `solo` command can't be found after installing.

Running this as administrator does not feel particularly good.

This is currently not a very user-friendly state, tbh.

ccoenen, 2022-02-16:
oh, well...
[image]