solokeys/solo1 #537: Updating a Hacker to a Secure Solo issue

AlexanderSchoenfeld, 2021-03-20:
Hi, 

I tried to change the hacker version to a secure version and followed this guide:
https://docs.solokeys.io/programming/

Here is the output I got. How should I continue?
```
c:\Temp>solo program aux enter-bootloader
Not using FIDO2 interface.
Solo rebooted.  Reconnecting...
Not using FIDO2 interface.

c:\Temp>solo program aux enter-dfu
Not using FIDO2 interface.
Please powercycle the device (pull out, plug in again)  <- I pulled it out and plugged it in again like it sayed

c:\Temp>solo program dfu bundle-secure-non-solokeys-4.1.1.hex
Traceback (most recent call last):
  File "c:\users\alexa\appdata\local\programs\python\python39\lib\runpy.py", line 197, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "c:\users\alexa\appdata\local\programs\python\python39\lib\runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "C:\Users\alexa\AppData\Local\Programs\Python\Python39\Scripts\solo.exe\__main__.py", line 7, in <module>
  File "c:\users\alexa\appdata\local\programs\python\python39\lib\site-packages\click\core.py", line 829, in __call__
    return self.main(*args, **kwargs)
  File "c:\users\alexa\appdata\local\programs\python\python39\lib\site-packages\click\core.py", line 782, in main
    rv = self.invoke(ctx)
  File "c:\users\alexa\appdata\local\programs\python\python39\lib\site-packages\click\core.py", line 1259, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "c:\users\alexa\appdata\local\programs\python\python39\lib\site-packages\click\core.py", line 1259, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "c:\users\alexa\appdata\local\programs\python\python39\lib\site-packages\click\core.py", line 1066, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "c:\users\alexa\appdata\local\programs\python\python39\lib\site-packages\click\core.py", line 610, in invoke
    return callback(*args, **kwargs)
  File "c:\users\alexa\appdata\local\programs\python\python39\lib\site-packages\solo\cli\program.py", line 63, in dfu
    dfu = solo.dfu.find(serial, attempts=connect_attempts)
  File "c:\users\alexa\appdata\local\programs\python\python39\lib\site-packages\solo\dfu.py", line 31, in find
    dfu.find(ser=dfu_serial, dev=raw_device, altsetting=altsetting)
  File "c:\users\alexa\appdata\local\programs\python\python39\lib\site-packages\solo\dfu.py", line 99, in find
    usb.core.find(idVendor=0x0483, idProduct=0xDF11, find_all=True)
  File "c:\users\alexa\appdata\local\programs\python\python39\lib\site-packages\usb\core.py", line 1299, in find
    raise NoBackendError('No backend available')
usb.core.NoBackendError: No backend available
```

AlexanderSchoenfeld, 2021-03-20:
After some more searching I found this issue:
https://github.com/solokeys/solo/issues/360

As kuba86 stated "Windows users need to install libusb for solo-python to work with Solo's DFU.", so I executed his steps until step 8. If I run step 9 I get now another error:

```
C:\Temp>solo program dfu bundle-secure-non-solokeys-4.1.1.hex
Traceback (most recent call last):
  File "c:\users\alexa\appdata\local\programs\python\python39\lib\runpy.py", line 197, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "c:\users\alexa\appdata\local\programs\python\python39\lib\runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "C:\Users\alexa\AppData\Local\Programs\Python\Python39\Scripts\solo.exe\__main__.py", line 7, in <module>
  File "c:\users\alexa\appdata\local\programs\python\python39\lib\site-packages\click\core.py", line 829, in __call__
    return self.main(*args, **kwargs)
  File "c:\users\alexa\appdata\local\programs\python\python39\lib\site-packages\click\core.py", line 782, in main
    rv = self.invoke(ctx)
  File "c:\users\alexa\appdata\local\programs\python\python39\lib\site-packages\click\core.py", line 1259, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "c:\users\alexa\appdata\local\programs\python\python39\lib\site-packages\click\core.py", line 1259, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "c:\users\alexa\appdata\local\programs\python\python39\lib\site-packages\click\core.py", line 1066, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "c:\users\alexa\appdata\local\programs\python\python39\lib\site-packages\click\core.py", line 610, in invoke
    return callback(*args, **kwargs)
  File "c:\users\alexa\appdata\local\programs\python\python39\lib\site-packages\solo\cli\program.py", line 63, in dfu
    dfu = solo.dfu.find(serial, attempts=connect_attempts)
  File "c:\users\alexa\appdata\local\programs\python\python39\lib\site-packages\solo\dfu.py", line 31, in find
    dfu.find(ser=dfu_serial, dev=raw_device, altsetting=altsetting)
  File "c:\users\alexa\appdata\local\programs\python\python39\lib\site-packages\solo\dfu.py", line 109, in find
    self.dev.set_configuration()
  File "c:\users\alexa\appdata\local\programs\python\python39\lib\site-packages\usb\core.py", line 905, in set_configuration
    self._ctx.managed_set_configuration(self, configuration)
  File "c:\users\alexa\appdata\local\programs\python\python39\lib\site-packages\usb\core.py", line 113, in wrapper
    return f(self, *args, **kwargs)
  File "c:\users\alexa\appdata\local\programs\python\python39\lib\site-packages\usb\core.py", line 158, in managed_set_configuration
    self.managed_open()
  File "c:\users\alexa\appdata\local\programs\python\python39\lib\site-packages\usb\core.py", line 113, in wrapper
    return f(self, *args, **kwargs)
  File "c:\users\alexa\appdata\local\programs\python\python39\lib\site-packages\usb\core.py", line 131, in managed_open
    self.handle = self.backend.open_device(self.dev)
  File "c:\users\alexa\appdata\local\programs\python\python39\lib\site-packages\usb\backend\libusb1.py", line 804, in open_device
    return _DeviceHandle(dev)
  File "c:\users\alexa\appdata\local\programs\python\python39\lib\site-packages\usb\backend\libusb1.py", line 652, in __init__
    _check(_lib.libusb_open(self.devid, byref(self.handle)))
  File "c:\users\alexa\appdata\local\programs\python\python39\lib\site-packages\usb\backend\libusb1.py", line 604, in _check
    raise USBError(_strerror(ret), ret, _libusb_errno[ret])
usb.core.USBError: [Errno 13] Access denied (insufficient permissions)
```

kuba86, 2021-03-21:
Hey, at the at of the traceback you can see: `usb.core.USBError: [Errno 13] Access denied (insufficient permissions)`
Are you running with Admin rights?

AlexanderSchoenfeld, 2021-03-21:
> Hey, at the at of the traceback you can see: `usb.core.USBError: [Errno 13] Access denied (insufficient permissions)`
> Are you running with Admin rights?

Yes, the command line is running with admin rights.

AlexanderSchoenfeld, 2021-03-21:
I downloaded the STM32CubeProgrammer and took a look at it.
Since I installed it the console program now shows another error:

```
C:\Temp>solo program dfu bundle-secure-non-solokeys-4.1.1.hex
erasing...
Traceback (most recent call last):
  File "c:\users\alexa\appdata\local\programs\python\python39\lib\runpy.py", line 197, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "c:\users\alexa\appdata\local\programs\python\python39\lib\runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "C:\Users\alexa\AppData\Local\Programs\Python\Python39\Scripts\solo.exe\__main__.py", line 7, in <module>
  File "c:\users\alexa\appdata\local\programs\python\python39\lib\site-packages\click\core.py", line 829, in __call__
    return self.main(*args, **kwargs)
  File "c:\users\alexa\appdata\local\programs\python\python39\lib\site-packages\click\core.py", line 782, in main
    rv = self.invoke(ctx)
  File "c:\users\alexa\appdata\local\programs\python\python39\lib\site-packages\click\core.py", line 1259, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "c:\users\alexa\appdata\local\programs\python\python39\lib\site-packages\click\core.py", line 1259, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "c:\users\alexa\appdata\local\programs\python\python39\lib\site-packages\click\core.py", line 1066, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "c:\users\alexa\appdata\local\programs\python\python39\lib\site-packages\click\core.py", line 610, in invoke
    return callback(*args, **kwargs)
  File "c:\users\alexa\appdata\local\programs\python\python39\lib\site-packages\solo\cli\program.py", line 99, in dfu
    dfu.write_page(i, data)
  File "c:\users\alexa\appdata\local\programs\python\python39\lib\site-packages\solo\dfu.py", line 211, in write_page
    assert DFU.state.DOWNLOAD_IDLE == self.state()
AssertionError
```

This is really frustraiting. If I knew it would be this hard on a Windows system I would never have tried it.
Also the documentation should state this part BEFORE showing the commands:

> Windows users need to install libusb for solo-python to work with Solo's DFU.

Could someone tell me how to update the Somu with the STM32CubeProgrammer?