solokeys/solo1 #539: Error Writing Seed from DiceKey

wolfewa, 2021-03-23:
Getting an Error 
Failed to wrote key due to unforeseen error: command=63, error=01

Just running Seed Solo 3.1.2 on the android version of the DiceKey app.  Not clear on what is generating the error.

nickgammon, 2021-05-27:
I got this exact error on my Solo 3.1.2 (Solo Tap USB-A), however I can successfully seed the Somu USB-A (hacker) version 5.0.0, and the Solu USB-C version 5.0.0.

Perhaps it is a firmware issue?

Mozgosup, 2021-06-23:
Same propblem here. Bought Solo Tap USB-A with my DiceKey, couldn't seed it. Neither with my Android app, nor with Windows command-line tool - same error 01.

conorpp, 2021-06-28:
Solo 3.x.y does not support DiceKeys, you need to use the DiceKeys fork of Solo (5.0.0 currently).  
@Mozgosup please check your firmware version.

The fork is here: https://github.com/conorpp/solo-dicekeys

And programming instructions here: https://github.com/conorpp/solo-dicekeys/releases/tag/5.0.0

RocketWolf2020, 2022-02-04:
Wait, whoa what? Forks, spoons, knives, linux, programming? I bought a Somu that cant be seeded (using my phone of course because it uses the camera to read the key) and it produces this error when trying to seed the somu. And now we are expecting customers to know programming and forks and what? I just want something that I paid for to work, not buy something that is broken and then learn programming and magic tricks and be expected to understand any of the tech talk going on...forking insane. I have been waiting **months** for just an acknowledgment to my queries for a replacement with something that works so that I dont feel like I threw away a bunch of money on a bunch of dice that become useless because the electronic device that uses it is broken and worthless. And thank you very much @BIzZmarCk [BIzZmarCk](https://github.com/BIzZmarCk) for bringing this post to my attention. At least someone is watching out for my interests and I wish I had sent you the money instead.

RussNelson, 2022-02-05:
It does seem MORE than a little bit strange that when you order both a SoloKey and a DiceKey set from Circuit Supply, you end up with a SoloKey that is incompatible with the DiceKey software.

Why would any sane person allow that to happen?

RussNelson, 2022-02-05:
Annnddddd.... it looks like I can't reprogram my SoloKey. How do I get one that is pre-programmed with the 5.0.0 version?

usb.core.USBError: [Errno 13] Access denied (insufficient permissions)

RussNelson, 2022-02-05:
Annndddddd .... my SoloKey has disappeared, replaced by the STM32 BOOTLOADER, with a difference idProduct of df11, and I can't talk to it using solo anymore.

RussNelson, 2022-02-11:
Anybody know how to get my SoloKey Tap USB-A out of dfu / bootloader mode? Yeah, I didn't read carefully and didn't see that the instructions were for a Solo Hacker. Very likely the SoloKey Tap has been permanently programmed, but you'd think that then I wouldn't be able to get it into dfu or even bootloader mode.

RussNelson, 2022-03-01:
Here's how I got it out of DFU mode: https://github.com/solokeys/solo/issues/601
I still can't reprogram it with the DiceKeys fork.

RocketWolf2020, 2022-03-03:
[image]
well, its been almost a full year now, and not a single response from the devs, customer service, or anyone. Not an single interest in helping resolve this issue, replacing the device or anything. If the devs want to make money "as a company" providing a product to the public, they got a LOT to learn. right now the total lack of interest, response, or any communication tells the world to "look the other way" when they think about purchasing anything.

nickray, 2022-03-05:
Conor answered this on [Jun 29, 2021](https://github.com/solokeys/solo/issues/539#issuecomment-870087134), the DiceKeys firmware is a separate experimental branch.

RussNelson, 2022-03-05:
Right, but how do I program this firmware into my Solo?