solokeys/solo1 #542: Makefile venv not in PHONY

pmassolino, 2021-04-22:
Hi,

I was following the building solutions and when I performed a make venv it gave me an error that venv was not installed. I proceed to installing it, but when I tried to perform make venv again, make would say it was done. I added venv in the .PHONY list of the Makefile and it fixed the problem. It would be good to add venv in the .PHONY list

nickray, 2021-04-22:
This doesn't sound right?
`make venv` creates the subdirectory `venv` with the virtualenv. The other targets have that as prerequisite.

pmassolino, 2021-04-22:
The machine used is a Ubuntu 20-04 and I didn't had installed the python3-venv package from apt. For some reason python created the folder venv, but crashed for not having venv installed. After installing venv from apt, I was not able to compile because the folder was created but empty.

enrikb, 2021-04-22:
Seems to be a Debianism. `python -m venv venv` requires python3-venv installed, unless called with `--without-pip`.
The half failing venv creation actually leaves a half installed `venv` directory behind. Fortunately, it returns failure and provides a helpful error message.

The make rules could be replaced by something like
```
 venv:
       python3 -m venv venv && \
       venv/bin/pip install -U pip || \
       rm -rf venv
```
to avoid this annoyance.

Of course, the half ready venv could also be fixed by `make clean venv` or `make -B venv` (gnu make).