solokeys/solo1 #544: Key will not enter dfu-mode

tspr, 2021-04-29:
current arch linux, i have 2 new solokeys from the dicekeys campaign. I wanted to re-flash the newest firmware, i get an error whe trying to put the key into dfu-mode:

 source ./bin/activate.fish                                                            755ms  Do 29 Apr 2021 07:51:57 CEST
(solokeys-dicekeys)  ◰³ solokeys-dicekeys  ~/w/solokeys-dicekeys  solo program aux enter-bootloader                      Do 29 Apr 2021 07:52:11 CEST
Solo rebooted.  Reconnecting...
(solokeys-dicekeys)  ◰³ solokeys-dicekeys  ~/w/solokeys-dicekeys  solo program aux enter-dfu                    1942ms  Do 29 Apr 2021 07:52:16 CEST
Traceback (most recent call last):
  File "/home/ts/workspace/solokeys-dicekeys/bin/solo", line 8, in <module>
    sys.exit(solo_cli())
  File "/home/ts/workspace/solokeys-dicekeys/lib/python3.9/site-packages/click/core.py", line 829, in __call__
    return self.main(*args, **kwargs)
  File "/home/ts/workspace/solokeys-dicekeys/lib/python3.9/site-packages/click/core.py", line 782, in main
    rv = self.invoke(ctx)
  File "/home/ts/workspace/solokeys-dicekeys/lib/python3.9/site-packages/click/core.py", line 1259, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/ts/workspace/solokeys-dicekeys/lib/python3.9/site-packages/click/core.py", line 1259, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/ts/workspace/solokeys-dicekeys/lib/python3.9/site-packages/click/core.py", line 1259, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/ts/workspace/solokeys-dicekeys/lib/python3.9/site-packages/click/core.py", line 1066, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/ts/workspace/solokeys-dicekeys/lib/python3.9/site-packages/click/core.py", line 610, in invoke
    return callback(*args, **kwargs)
  File "/home/ts/workspace/solokeys-dicekeys/lib/python3.9/site-packages/solo/cli/program.py", line 248, in enter_dfu
    p.enter_st_dfu()
  File "/home/ts/workspace/solokeys-dicekeys/lib/python3.9/site-packages/solo/devices/solo_v1.py", line 234, in enter_st_dfu
    self.send_only_hid(SoloBootloader.CommandBoot, req)
  File "/home/ts/workspace/solokeys-dicekeys/lib/python3.9/site-packages/solo/devices/solo_v1.py", line 108, in send_only_hid
    self.dev._dev.InternalSend(0x80 | cmd, bytearray(data))
AttributeError: 'CtapHidDevice' object has no attribute '_dev'

tspr, 2021-04-29:
uname -a                                   341ms  Do 29 Apr 2021 07:52:23 CEST
Linux ****** 5.11.16-zen1-1-zen #1 ZEN SMP PREEMPT Wed, 21 Apr 2021 17:22:09 +0000 x86_64 GNU/Linux

enrikb, 2021-04-29:
This is due to the recent change to python-fido2 0.9.1 (solo-python 0.0.28 and newer).
You could try to use solo-python 0.0.27 and check if it still works for you.
Alternatively, you can try my very experimental fix for the issue here: enrikb/solo-python@639f44872df5e4c93844c678f10a522ff4eeea4b

conorpp, 2021-04-29:
I just released a solo-python 0.0.30 which should solve this.  Please update solo-python and try again.

tspr, 2021-04-29:
Hi Enrik and Conor,
superfast replies. Awesome! Tried 0.0.30 and it worked! Thanks!

closed 2021-04-29 by tspr