solokeys/solo1 #546: Explain how solokey works: It is a (usb) keyboard emulator?

reikred, 2021-05-02:
I find it amazing that after 2 hours of reading about solokey, yubikey, etc, it was only by accident that I discovered somewhere a on the web a hint as to what the key actually DOES (and now I can't even find that hint again ;))

THE ESSENCE: Solokey (and other security keys) are (usb) KEYBOARD EMULATORS. They are devices that TYPE in passwords (or OAuths or similar) for you, at the push of a button or some such manual action.

That is the essence of solokey, unless I am mistaken. There must of course be much more to it than that, such as (a) the key being able to check that the entity requesting a password really is who they claim to be (b) that the solokey itself is encrypted such that keys cannot be stolen, and (c) many more details and layers of complexity. But to leave out the very essence of the solokey hardware from the README.md and other docs is mystifying.

Maybe I have misunderstood how solokey works. But in any case someone here can perhaps figure out how and where to explain prominently how the system works?

nickray, 2021-05-02:
No, they're roaming FIDO authenticators. Maybe https://webauthn.guide/ is the best entry point. The point of WebAuthn/FIDO being to get rid of passwords.

reikred, 2021-05-02:
Is it true or false that the manner of the transfer of the {AUTHn, OAuth, password, something} secret access token (or whatever you want to call it in any given case) takes place by the solokey emulating a keyboard and sending the token as USB serial data that ends up in a keyboard buffer at the host?