solokeys/solo1 #547: Solo Key not recognized

lambaradan, 2021-05-06:
Hello. I was redirected here by Haden at hello@solokeys.com.

I have just received a Solo Key USB-A via DiceKeys. I used your Python command line app to update the firmware, and I think this might have been a mistake. It was v. 5.0 something when I got it, and when upgraded, it was upgraded (??) to version 4.1.2. Then I had trouble in the common line app adding a credential...

I then found your GitHub tutorial on DiceKeys Seeding WebAuthN Authenticator Credentials and tried to run that, but now the Solo Key is not recognized.

How can I reset it totally in a mode that it will with with the DiceKeys app? Holding down the button while inserting the key and waiting for the blinking light produces no light.

The Solo Key is not recognized by the DiceKeys app, or your python command line program. The error message is: "solo.exceptions.NoSoloFoundError: no Solo found" in response to a "solo key verify" command.

However, on my Mac Mini in System Information, the hardware is detected:
STM32  BOOTLOADER:

  Product ID: 0xdf11
  Vendor ID: 0x0483  (STMicroelectronics)
  Version: 22.00
  Serial Number: 208F31A15653
  Speed: Up to 12 Mb/s
  Manufacturer: STMicroelectronics
  Location ID: 0x14300000 / 28
  Current Available (mA): 500
  Current Required (mA): 100
  Extra Operating Current (mA): 0

Please help.

conorpp, 2021-05-06:
It looks like it is in DFU mode.  

Run these two commands:

```
# program firmware
solo program dfu bundle-hacker-5.0.0.hex
```

```
# reboot
solo program aux leave-dfu
```

Do not run the later command if the first command doesn't succeed.

https://github.com/conorpp/solo-dicekeys/releases/tag/5.0.0

lambaradan, 2021-05-06:
@conorpp 
No dice.

Mac-mini:~ minireduxx$ solo program dfu bundle-hacker-5.0.0.hex
Traceback (most recent call last):
  File "/usr/local/bin/solo", line 8, in <module>
    sys.exit(solo_cli())
  File "/usr/local/lib/python3.9/site-packages/click/core.py", line 829, in __call__
    return self.main(*args, **kwargs)
  File "/usr/local/lib/python3.9/site-packages/click/core.py", line 782, in main
    rv = self.invoke(ctx)
  File "/usr/local/lib/python3.9/site-packages/click/core.py", line 1259, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/local/lib/python3.9/site-packages/click/core.py", line 1259, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/local/lib/python3.9/site-packages/click/core.py", line 1066, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/local/lib/python3.9/site-packages/click/core.py", line 610, in invoke
    return callback(*args, **kwargs)
  File "/usr/local/lib/python3.9/site-packages/solo/cli/program.py", line 63, in dfu
    dfu = solo.dfu.find(serial, attempts=connect_attempts)
  File "/usr/local/lib/python3.9/site-packages/solo/dfu.py", line 31, in find
    dfu.find(ser=dfu_serial, dev=raw_device, altsetting=altsetting)
  File "/usr/local/lib/python3.9/site-packages/solo/dfu.py", line 101, in find
    usb.core.find(idVendor=0x0483, idProduct=0xDF11, find_all=True)
  File "/usr/local/lib/python3.9/site-packages/usb/core.py", line 1299, in find
    raise NoBackendError('No backend available')
usb.core.NoBackendError: No backend available
Mac-mini:~ minireduxx$

conorpp, 2021-05-06:
Can you try again after installing libusb

```
brew install libusb libusb-compat 
```

lambaradan, 2021-05-06:
libusb and libusb-compat successfully installed.

However:

Mac-mini:~ minireduxx$ solo program dfu bundle-hacker-5.0.0.hex
Traceback (most recent call last):
  File "/usr/local/bin/solo", line 8, in <module>
    sys.exit(solo_cli())
  File "/usr/local/lib/python3.9/site-packages/click/core.py", line 829, in __call__
    return self.main(*args, **kwargs)
  File "/usr/local/lib/python3.9/site-packages/click/core.py", line 782, in main
    rv = self.invoke(ctx)
  File "/usr/local/lib/python3.9/site-packages/click/core.py", line 1259, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/local/lib/python3.9/site-packages/click/core.py", line 1259, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/local/lib/python3.9/site-packages/click/core.py", line 1066, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/local/lib/python3.9/site-packages/click/core.py", line 610, in invoke
    return callback(*args, **kwargs)
  File "/usr/local/lib/python3.9/site-packages/solo/cli/program.py", line 77, in dfu
    ih.fromfile(firmware, format="hex")
  File "/usr/local/lib/python3.9/site-packages/intelhex/__init__.py", line 249, in loadfile
    self.loadhex(fobj)
  File "/usr/local/lib/python3.9/site-packages/intelhex/__init__.py", line 199, in loadhex
    fobj = open(fobj, "r")
FileNotFoundError: [Errno 2] No such file or directory: 'bundle-hacker-5.0.0.hex'
Mac-mini:~ minireduxx$

conorpp, 2021-05-06:
For this you will need to download the firmware file and provide the correct path.

Download: https://github.com/conorpp/solo-dicekeys/releases/download/5.0.0/bundle-secure-non-solokeys-5.0.0.hex

cd to your downloads folder and run the command again.

lambaradan, 2021-05-06:
Success! Solid green light now. Is the error: Warning: OSError with write_page" anything to worry about?
Key is recognized by the DiceKeys app and was successfully seeded.

A big thank you Mr. Conor!

Mac-mini:Downloads minireduxx$ solo program dfu bundle-hacker-5.0.0.hex 
erasing...
downloading 100.00%  08038800 - 0803882e ...         
time: 5951 ms
verifying...
reading 100.00%  08038800 - 0803882e ...         
firmware readback verified.
Mac-mini:Downloads minireduxx$ solo program aux leave-dfu
Rewriting option bytes...
Warning: OSError with write_page
Please powercycle the device (pull out, plug in again)
Mac-mini:Downloads minireduxx$

closed 2021-05-06 by lambaradan