solokeys/solo1 #551: Errors when trying to build the firmware

s-schoen, 2021-05-29:
I tried to compile the source code for the use without hardware, but i am getting some errors and i am not sure why.

Setup
- Ubuntu 21.04
- Installed arm tool chain from [https://developer.arm.com/open-source/gnu-toolchain/gnu-rm/downloads](https://developer.arm.com/open-source/gnu-toolchain/gnu-rm/downloads)
- Installed `libsodium-dev`
- Installed `solo-python`

Compiling the source code according to the readme
```shell
git clone --recurse-submodules https://github.com/solokeys/solo
cd solo
make all
```
results in the following errors

```
[...]

bootloader/bootloader.c:23:28: error: expected expression before ',' token
   23 |   .major = SOLO_VERSION_MAJ,
      |                            ^
bootloader/bootloader.c: In function 'bootloader_bridge':
bootloader/bootloader.c:264:39: error: expected expression before ';' token
  264 |             version = SOLO_VERSION_MAJ;
      |                                       ^
bootloader/bootloader.c:266:39: error: expected expression before ';' token
  266 |             version = SOLO_VERSION_MIN;
      |                                       ^
arm-none-eabi-gcc src/led.c -mcpu=cortex-m4 -mfpu=fpv4-sp-d16 -mfloat-abi=hard -mthumb  -Os -Ibootloader/ -Isrc/ -Isrc/cmsis/ -Ilib/ -Ilib/usbd/ -I../../fido2/ -I../../fido2/extensions -I../../tinycbor/src -I../../crypto/sha256 -I../../crypto/micro-ecc -I../../crypto/tiny-AES-c -I../../crypto/cifra/src -I../../crypto/cifra/src/ext -I../../crypto/salty/c-api -c -DDEBUG_LEVEL=0 -DSTM32L432xx -DAES256=1  -DUSE_FULL_LL_DRIVER -DAPP_CONFIG=\"bootloader.h\" -DSOLO_HACKER   -Wall -Wextra -Wno-unused-parameter -Wno-missing-field-initializers -fdata-sections -ffunction-sections -mcpu=cortex-m4 -mfpu=fpv4-sp-d16 -mfloat-abi=hard -mthumb -g -DSOLO_VERSION_MAJ= -DSOLO_VERSION_MIN= -DSOLO_VERSION_PATCH= -DSOLO_VERSION=\"4.1.2-2-g2884f95\" -o src/led.o
bootloader/bootloader.c:268:41: error: expected expression before ';' token
  268 |             version = SOLO_VERSION_PATCH;

[...]
```
Same issue occurs when trying to build the firmware as described here [https://docs.solokeys.dev/building/](https://docs.solokeys.dev/building/)

Did i miss some dependency or any critical step?

svn2208, 2022-02-10:
For me I could solve it my linking the python3 to python 
`/usr/bin/python -> python3`

nickray, 2022-03-06:
I wonder if these platforms will ever finally link python to the non-deprecated version...