solokeys/solo1 #552: Unable to read solo hacker output using monitor

sravya-du, 2021-06-07:
I'm trying to read the output of my solo hacker using the monitor tool (./solo monitor). I'm using a mac and a USB-C solo hacker. It just stays hung on "reconnecting..." for all of the serial ports I've tried. In addition, when my solo key is plugged in (the green light is turned on), I don't see any result containing "usb" when running the command "ls /dev/tty.*". Is there an issue with detecting/connecting with the solo key on macs?

enrikb, 2021-06-07:
Have you installed a debug-enabled firmware? (`make firmware-debug-1` or `make firmware-debug-2`, see https://docs.solokeys.dev/building/).

sravya-du, 2021-06-07:
I did, but I didn't run make clean so I think the wrong version of the firmware was uploaded. Tried again with make clean first and it worked. Thanks!

On another note, is there any documentation on running the solo hacker code without a device and sending commands to it? when I run "./solo ls", it only detects when I have physical solokeys inserted