solokeys/solo1 #562: Solo Not Found: Windows detects it as STM32 BOOTLOADER

isaacwalkercox, 2021-07-03:
Hi there,

I bought a Solo Key 2.1 version from Amazon and it is not being detected by the solo-python package when using solo ls or solo key wink. Windows says that it finds not driver and cannot install the device when I click on the above device under Other devices in device manager.

Thank you.

isaacwalkercox, 2021-07-04:
After further research, it seems this is the default firmware of the chip that the Solo is using. Is it possible that the firmware just wasn't flashed before shipping on the TWO models I bought from the same supplier on Amazon?

davidjw1979, 2021-07-04:
@isaacwalkercox 

Firstly I'm not anything to do with the project.  I'm also in the same boat (emailed the hello@solokeys.com address; still waiting for a response).

Also through digging around on Github I've found others with the same issue (see #550 amonst others) so I'm guessing that there is a batch problem.

However I was able (and on a Windows PC) to get the firmware, and as such the device, working as it should have arrived.

It required me installing Python for Windows, Wget.exe and using Zadig to load in libusb (as well as downloading the firmware from here and modifying my Windows path).  

I'm just off to work so don't have time to type up everthing I did, but if you want I can write up what I did later; it's not for a PC begginer though.

isaacwalkercox, 2021-07-05:
> @isaacwalkercox
> 
> Firstly I'm not anything to do with the project. I'm also in the same boat (emailed the [hello@solokeys.com](mailto:hello@solokeys.com) address; still waiting for a response).
> 
> Also through digging around on Github I've found others with the same issue (see #550 amonst others) so I'm guessing that there is a batch problem.
> 
> However I was able (and on a Windows PC) to get the firmware, and as such the device, working as it should have arrived.
> 
> It required me installing Python for Windows, Wget.exe and using Zadig to load in libusb (as well as downloading the firmware from here and modifying my Windows path).
> 
> I'm just off to work so don't have time to type up everything I did, but if you want I can write up what I did later; it's not for a PC begginer though.

Thanks for the reply. I am guessing you flashed a DFU update to the device in some way? That was going to be my next step. I would appreciate it if we could work together to fix this problem once and for all. 

Thank you,

Isaac.

isaacwalkercox, 2021-07-05:
I checked #550 to realize I was correct in my thinking. I will try to flash the firmware via DFU and, for the sake of documenting the process, keep you informed. Thank you.

davidjw1979, 2021-07-05:
I was mostly responding to ensure you didn't feel like you were shouting into thin air and that it was something that others seem to have had issues with - myself included.

Taken from #525, my hazy memory, and other searches this is roughly what I did:

Installed python from https://www.python.org/downloads/windows/
Added 'C:\Users\yourusernamegoeshere\AppData\Local\Programs\Python\Python39\Scripts' to my path

installed wget.exe from https://eternallybored.org/misc/wget/
moved wget into windows\system32

Used zadig https://zadig.akeo.ie/ to install libusb (taken from https://github.com/solokeys/solo-python#readme)

in a cmd window (that had admin rights)

'pip install solo-python

wget https://github.com/solokeys/solo/releases/download/4.1.1/bundle-secure-non-solokeys-4.1.1.hex

solo program dfu bundle-secure-non-solokeys-4.1.1.hex
solo program aux leave-dfu

solo key wink
solo key update
solo key reset'

Which resulted in it working (I think!).
(N.B some of the commands may not be needed, and I can't vouch for what I downloaded)

isaacwalkercox, 2021-07-05:
> I was mostly responding to ensure you didn't feel like you were shouting into thin air and that it was something that others seem to have had issues with - myself included.
>
> Taken from #525, my hazy [...]

Having run the first few DFU commands the USB is now not being detected in either normal (user-mode) or DFU mode... I think I broke it when I was using a separate utility to try and flash the firmware because it exposes 4 different DFU interfaces and I flashed to each one, again, this was BEFORE running the DFU commands above, which had seemingly worked successfully. I am very confused.

isaacwalkercox, 2021-07-05:
I have decided to return the second Solo Key I bought from Amazon. I have had an interesting time trying to get it to work but ultimately I need something that works now, not after I have had to fiddle around with it, along with other reasons. Thank you for your help though.

closed 2021-07-05 by isaacwalkercox

noteknix, 2021-12-27:
Tried the instructions above.  Looked OK until solo key update.  
[image]

I now have a pair of keys with disco lights when plugged in! :-)  CTAP errors if you try any commands and I can't even find what they're appearing as in Device Manager.