solokeys/solo1 #565: FIDO MDS V3

MichaelGrafnetter, 2021-07-30:
Hi, I wonder why *Solo FIDO2 Authenticator* is present in the now legacy [FIDO Metadata Service 2](https://fidoalliance.org/legacy-mds2-information/), but is missing from the current [FIDO Metadata Service 3](https://fidoalliance.org/metadata/). Has Solo lost the certification? See the cool [FIDO MDS Explorer](https://opotonniee.github.io/fido-mds-explorer/) for an up-to-date list of certified devices.

My1, 2021-11-14:
> As part of this I'd also suggest to fill U2F records and not just FIDO2 records as those are used when for example Firefox on Linux is used.

I'd assume android as well, as that cannot do CTAP2

jtalir, 2022-07-12:
Still no news?

aseigler, 2022-08-18:
It's in there now!

jtalir, 2022-08-23:
Great, I see records for FIDO2 protocol. Would be possible to add also records for U2F protocol so it can be detected also by systems that do not speak CTAP2 (Firefox on Linux and Mac and also by Android phones)?