solokeys/solo1 #570: Cannot list or create credentials on SoloKey with no pin set on the key

filipkemuel, 2021-09-05:
When I try to create or list credentials on my SoloKey it ask me to type a pin.
I have no pin set on the key so I leave it blank and just press enter. 
The program then fails.

> Firmware version: 4.1.2 locked
> solo-python version: 0.0.30
> 

```
Traceback (most recent call last):
  File "/usr/bin/solo", line 8, in <module>
    sys.exit(solo_cli())
  File "/usr/lib/python3.9/site-packages/click/core.py", line 1137, in __call__
    return self.main(*args, **kwargs)
  File "/usr/lib/python3.9/site-packages/click/core.py", line 1062, in main
    rv = self.invoke(ctx)
  File "/usr/lib/python3.9/site-packages/click/core.py", line 1668, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/lib/python3.9/site-packages/click/core.py", line 1668, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/lib/python3.9/site-packages/click/core.py", line 1668, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/lib/python3.9/site-packages/click/core.py", line 1404, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/lib/python3.9/site-packages/click/core.py", line 763, in invoke
    return __callback(*args, **kwargs)
  File "/usr/lib/python3.9/site-packages/solo/cli/key.py", line 553, in cred_ls
    cm = client.cred_mgmt(pin)
  File "/usr/lib/python3.9/site-packages/solo/devices/base.py", line 116, in cred_mgmt
    token = client.client_pin.get_pin_token(pin)
  File "/usr/lib/python3.9/site-packages/fido2/ctap2/pin.py", line 256, in get_pin_token
    resp = self.ctap.client_pin(
  File "/usr/lib/python3.9/site-packages/fido2/ctap2/base.py", line 723, in client_pin
    return self.send_cbor(
  File "/usr/lib/python3.9/site-packages/fido2/ctap2/base.py", line 675, in send_cbor
    raise CtapError(status)
fido2.ctap.CtapError: CTAP error: 0x35 - PIN_NOT_SET
[1]    25362 exit 1     solo key credential ls
```

```
Traceback (most recent call last):
  File "/usr/bin/solo", line 8, in <module>
    sys.exit(solo_cli())
  File "/usr/lib/python3.9/site-packages/click/core.py", line 1137, in __call__
    return self.main(*args, **kwargs)
  File "/usr/lib/python3.9/site-packages/click/core.py", line 1062, in main
    rv = self.invoke(ctx)
  File "/usr/lib/python3.9/site-packages/click/core.py", line 1668, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/lib/python3.9/site-packages/click/core.py", line 1668, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/lib/python3.9/site-packages/click/core.py", line 1404, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/lib/python3.9/site-packages/click/core.py", line 763, in invoke
    return __callback(*args, **kwargs)
  File "/usr/lib/python3.9/site-packages/solo/cli/key.py", line 151, in make_credential
    solo.hmac_secret.make_credential(
  File "/usr/lib/python3.9/site-packages/solo/hmac_secret.py", line 42, in make_credential
    attestation_object = client.make_credential(
  File "/usr/lib/python3.9/site-packages/fido2/client.py", line 446, in make_credential
    self._verify_rp_id(options.rp.id)
  File "/usr/lib/python3.9/site-packages/fido2/client.py", line 305, in _verify_rp_id
    raise ClientError.ERR.BAD_REQUEST()
fido2.client.ClientError: (<ERR.BAD_REQUEST: 2>, None)
[1]    25681 exit 1     solo key make-credential --host [censored] --user [censored]
```

msiqx9, 2021-09-05:
Set a pin.

filipkemuel, 2021-09-06:
I know setting a pin would make it work.. but then I would have to use the pin every time I use the key for 2FA on websites.. making it 3FA.. dince I would then need both password, key and pin... and that would be quite annoying..

Should it not be possible to use the ley without setting a pin?

nuno0529, 2021-09-07:
ctap2 spec allows to create credential w/o using UV (e.g PIN here) and I don't see any problem on the solo key I have (fw v4.0.0) with actual ctap2 command. I think the error code in your log points to this line, https://github.com/Yubico/python-fido2/blob/142587b3e698ca0e253c78d75758fda635cac51a/fido2/client.py#L305

But for cred_mgmt, ctap2 spec does require w/ UV(e.g. PIN here).