solokeys/solo1 #581: ERROR Failed to write key due to unforeseen error: command=63, error=01

RocketWolf2020, 2021-10-14:
Hi Team, I just received my discekey and somu today. everything is working great until I proceed to use the somu key (solo v2) to generate a seed. I plug in the somu and the green light comes on and everything looks good, I tap "Seed SOLO 3.1.2 S/N# xxxxxx" I get:
ERROR Failed to write key due to unforeseen error: command=63, error=01
And that's it, nothing else happens.
Samsung Galaxy A72 SM-A725F/DS mobile phone using USB-A/B to USB-C adapter, android 11, One UI v3.1, kernel 4,14,190, Knox v3.7 (Knox API level 33), Android Security Patch level 1 Sept 2021
Kind Regards,

RocketWolf2020, 2022-01-04:
any help at all would be appreciated, or a new key.

RocketWolf2020, 2022-02-02:
almost 4 months and not even an acknowledgment.

RocketWolf2020, 2022-02-02:
So I guess the email to hello@solokeyes was 100% pointless

SoloKeys USA <hello@solokeys.com> | SoloKeys USA <hello@solokeys.com> | Thu, Oct 14, 2021, 4:45 AM |   | Reply
-- | -- | -- | -- | --
SoloKeys USA <hello@solokeys.com>
to me 

Scott...please post technical issues here for support: https://github.com/solokeys/solo/issues.

BIzZmarCk, 2022-02-03:
Hey there im not too sure if you found any help for your questions but it's more ore less a double post : 
https://github.com/solokeys/solo/issues/539
and to resolve this issue (if I'm not mistaken, I'm not part of the solokeys team) you need to use this:
https://github.com/conorpp/solo-dicekeys/releases/tag/5.0.0
best regards BIzZmarCk

RocketWolf2020, 2022-02-04:
Thanks so much for catching that and making the friendly gesture to alert
me, much appreciated :)