solokeys/solo1 #584: Can't get into bootloader mode

rototom, 2021-11-07:
I try to upgrade my solo, but I can't get into bootloader mode. It does not flash yellow, when I press the button while plugging it in.
205833AD344B: SoloKeys Solo 2.4.3

maweil, 2021-11-07:
Have you tried to keep the button pressed for at least 2 more seconds after plugging in?

rototom, 2021-11-07:
Yes, I tried it a dozen times with varying timespans.

maweil, 2021-11-07:
> Yes, I tried it a dozen times with varying timespans.

Ok, that doesn't sound too well. I have a similarly old key and this method works reliably for years now. 
The only other method I know of would be running `solo program aux enter-bootloader` as listed in the [documentation](https://docs.solokeys.dev/bootloader-mode/). Unfortunately this method is designed for newer keys (> 2.5.3)

I tried that method with my key a few minutes ago. Unfortunately it did not start into bootloader automatically. What did work was plugging in the key (without pressing anything), starting to hold down the button and while doing so running the mentioned command. With that, my key entered the bootloader as well. As this method still requires using the button, I'm not sure if it will help you, though...

If that method does not work, I hope someone else can help you further.

rototom, 2021-11-26:
Simple trick is to push that button really (really) hard. Works now!