solokeys/solo1 #587: Solokeys gives read errors

montywi, 2021-11-11:
I just bought 3 solo keys, CS-SOMU-02 and I get read errors from all of them.
I am using OpenSuse 15.3 Leap , but I also tested on my old Windows machine and could not get them to work there either.

On Linux, I get this on the message log when inserting any of the keys:
2021-11-11T17:10:03.276085+02:00 storm kernel: [700040.913336] usb 1-2: new full-speed USB device number 39 using xhci_hcd
2021-11-11T17:10:03.404155+02:00 storm kernel: [700041.041370] usb 1-2: device descriptor read/64, error -71
2021-11-11T17:10:03.640154+02:00 storm kernel: [700041.277351] usb 1-2: device descriptor read/64, error -71
2021-11-11T17:10:03.876076+02:00 storm kernel: [700041.513336] usb 1-2: new full-speed USB device number 40 using xhci_hcd
2021-11-11T17:10:04.004154+02:00 storm kernel: [700041.641351] usb 1-2: device descriptor read/64, error -71
2021-11-11T17:10:04.240153+02:00 storm kernel: [700041.877353] usb 1-2: device descriptor read/64, error -71
2021-11-11T17:10:04.348154+02:00 storm kernel: [700041.985389] usb usb1-port2: attempt power cycle
2021-11-11T17:10:04.760162+02:00 storm kernel: [700042.397345] usb 1-2: new full-speed USB device number 41 using xhci_hcd
2021-11-11T17:10:04.760188+02:00 storm kernel: [700042.397542] usb 1-2: Device not responding to setup address.
2021-11-11T17:10:04.760190+02:00 storm kernel: [700042.397594] debugfs: Directory '11' with parent 'devices' already present!
2021-11-11T17:10:04.968156+02:00 storm kernel: [700042.605532] usb 1-2: Device not responding to setup address.
2021-11-11T17:10:04.968177+02:00 storm kernel: [700042.605578] debugfs: Directory '12' with parent 'devices' already present!
2021-11-11T17:10:05.176158+02:00 storm kernel: [700042.813337] usb 1-2: device not accepting address 41, error -71
2021-11-11T17:10:05.304157+02:00 storm kernel: [700042.941342] usb 1-2: new full-speed USB device number 42 using xhci_hcd
2021-11-11T17:10:05.304177+02:00 storm kernel: [700042.941538] usb 1-2: Device not responding to setup address.
2021-11-11T17:10:05.304179+02:00 storm kernel: [700042.941587] debugfs: Directory '14' with parent 'devices' already present!
2021-11-11T17:10:05.512156+02:00 storm kernel: [700043.149540] usb 1-2: Device not responding to setup address.
2021-11-11T17:10:05.512177+02:00 storm kernel: [700043.149585] debugfs: Directory '15' with parent 'devices' already present!
2021-11-11T17:10:05.720156+02:00 storm kernel: [700043.357341] usb 1-2: device not accepting address 42, error -71
2021-11-11T17:10:05.720176+02:00 storm kernel: [700043.357383] usb usb1-port2: unable to enumerate USB device

Any idea of how to continue or should I ask for a refund of the keys?

msiqx9, 2021-11-17:
You may have received un-programmed keys.  2 of the 3 keys I purchased via Amazon UK this year at different times have been un-programmed.  You can attempt programming as discussed in [#550](https://github.com/solokeys/solo/issues/550)

nickray, 2021-11-21:
@montywi better than attempt to program yourself is to send customer support an email to replace them (hello@solokeys.com).