solokeys/solo1 #588: Is the crypto hardware accelerated?

skannan89, 2021-11-22:
Hello,

Looks like the MCU supports hardware acceleration of certain crypto algorithms. Does solo support it?

Thanks!

coelner, 2021-11-26:
Solo v1: https://www.st.com/en/microcontrollers-microprocessors/stm32l432kc.html
Solo v2: https://www.nxp.com/products/processors-and-microcontrollers/arm-microcontrollers/general-purpose-mcus/lpc5500-cortex-m33/high-efficiency-arm-cortex-m33-based-microcontroller-family:LPC55S6x

I did not see any embedded crypto for the solo v1.

nickray, 2021-11-27:
Correct, there is no acceleration on Solo 1.
On Solo 2 there is a crypto coprocessor, which is currently not used. It may be used in future firmware updates. At least for p256, our implementation is faster than the vendor's SDK example 😅.

closed 2022-04-24 by skannan89