solokeys/solo1 #589: wrong id being returned in ctap2 (webauthn) on chrome/chromium

flumm, 2021-12-06:
Hi,

i have a weird issue when i register multiple different keys (e.g. an OnlyKey and a SoloKey) on the same site.
When logging in, the solokey (and the onlyke for that matter) only return the first id thats provided in the allowCredentials portion of
the 'navigator.credentials.get' call

i read through the relevant ctap2 source code here, but did not find some obvious bug, maybe someone else can help me.

first, i thought it was a chrome/chromium bug (since it works on firefox), but it seems the wrong id already comes back from the
solokey (see https://bugs.chromium.org/p/chromium/issues/detail?id=1274509)

the log from  chromes 'chrome://device-log' is here:

After registering the OnlyKey first and then the SoloKey on 'webauthn.io' here is the
(successful) login with the OnlyKey:

```
FIDODebug[08:22:10] -> {1: {"id": h'473511DAAE38F6AA51ACCDE61986A4CFF6C7BDA6DAAADACABEAC12AA1E9FA04468703AD9EB66D9F23F3E195B9BFDC588', "type": "public-key"}, 2: h'74A6EA9213C99C2F74B22492B320CF40262A94C1A950A0397F29250B60841EF00500E4FFD4', 3: h'30440220357B73DE02F309D5898E3FE6B49F8965E34FCBD7F943C8040E8764D82B33358002203A9C6112077943D9E5BD2558DE07D6455A162EF711A87E743E57775E3508FA72'}
FIDODebug[08:22:09] <- 2 {1: "webauthn.io", 2: h'D431B35DBB91B341ACFC9F8F53BBA7FC80822A361A88858E3EAC8A92472270BB', 3: [{"id": h'473511DAAE38F6AA51ACCDE61986A4CFF6C7BDA6DAAADACABEAC12AA1E9FA04468703AD9EB66D9F23F3E195B9BFDC588', "type": "public-key"}, {"id": h'EBF0924BCE17BBD61C1092EC09E466A0ECFE91A89C30E9DD0C69FC019FA1429DB7D2756512AE9078498E237FC6B0395E', "type": "public-key"}], 6: h'4CC1BD4794D60B175BF7E20461C0E920', 7: 1}
FIDODebug[08:22:09] -> {2: h'F8DD5492A9A349BA4D5016185A64BA4B'}                 
FIDODebug[08:22:09] <- 6 {1: 1, 2: 5, 3: {1: 2, 3: -25, -1: 1, -2: h'CE481CB814D716455B09FF6C2A5726FBC6C7A9340A769BADB0D7E8C6E604E28B', -3: h'3978C1FB24436D488270CFC8CF8A8676679458AD2D64E65ED290EB4E92BC1F2C'}, 6: h'61B17D45565AE33B7EF8F5EB72AA467A'}
FIDODebug[08:22:09] -> {1: {1: 2, 3: -25, -1: 1, -2: h'987DECCC8114D0BC1657EAD4CEF439A3C848FE285C5663C5CFDA5139B4226393', -3: h'22B5E7BBBD7DD6F2041C10438391C2D535D1F991748A273EA6E762DD9BE40684'}}
FIDODebug[08:22:09] <- 6 {1: 1, 2: 2}                                           
FIDODebug[08:22:06] -> {3: 8}                                                   
FIDODebug[08:22:06] Ignoring status 45 from hid:d9499d64-75d4-4df6-8bf3-80a249674609
FIDODebug[08:22:06] -> (CTAP2 error code 45)                                    
FIDODebug[08:22:06] <- 6 {1: 1, 2: 1}                                           
FIDODebug[08:22:06] <- (cancel)                                                 
FIDODebug[08:22:06] -> (CTAP2 error code 51)                                    
FIDODebug[08:22:05] <- 1 {1: h'E3B0C44298FC1C149AFBF4C8996FB92427AE41E4649B934CA495991B7852B855', 2: {"id": ".dummy"}, 3: {"id": h'01', "name": "dummy"}, 4: [{"alg": -7, "type": "public-key"}], 8: h'', 9: 1}
FIDODebug[08:22:05] The device supports the CTAP2 protocol.                     
FIDODebug[08:22:05] Unexpected protocol version received.                       
FIDODebug[08:22:05] -> {1: ["U2F_V2", "FIDO_2_0", "FIDO_2_1_PRE"], 2: ["credProtect", "hmac-secret"], 3: h'79D699DF01914B10B9035467E7CE8231', 4: {"rk": true, "up": true, "plat": false, "credMgmt": true, "clientPin": true}, 5: 1200, 6: [1], 7: 20, 8: 256}
FIDODebug[08:22:05] <- 1 {1: h'E3B0C44298FC1C149AFBF4C8996FB92427AE41E4649B934CA495991B7852B855', 2: {"id": ".dummy"}, 3: {"id": h'01', "name": "dummy"}, 4: [{"alg": -7, "type": "public-key"}], 8: h'', 9: 1}
FIDODebug[08:22:05] The device supports the CTAP2 protocol.                     
FIDODebug[08:22:05] Unexpected protocol version received.                       
FIDODebug[08:22:05] -> {1: ["U2F_V2", "FIDO_2_0", "FIDO_2_1_PRE"], 2: ["credProtect", "hmac-secret"], 3: h'8976631BD4A0427F57730EC71C9E0279', 4: {"rk": true, "up": true, "plat": false, "credMgmt": true, "clientPin": true}, 5: 1200, 6: [1], 7: 20, 8: 128}
FIDODebug[08:22:05] Sending CTAP2 AuthenticatorGetInfo request to authenticator.
FIDOEvent[08:22:05] Starting GetAssertion flow      
```                            
                                                                                
And here is the (unsuccesful) login with the SoloKey:                                                             
                                                                                
```
FIDODebug[08:22:57] -> {1: {"id": h'473511DAAE38F6AA51ACCDE61986A4CFF6C7BDA6DAAADACABEAC12AA1E9FA04468703AD9EB66D9F23F3E195B9BFDC588', "type": "public-key"}, 2: h'74A6EA9213C99C2F74B22492B320CF40262A94C1A950A0397F29250B60841EF0050000069D', 3: h'3046022100EDF451D923203A531DD435B0CCD52C81A6C9B6A447A3B0D4F41E9F9D4AE444D5022100BB9C89C3404B1ECCA8983298A8CDFE3D662AB2E91F6144146B6F583BE069FFDC'}
FIDODebug[08:22:56] <- 2 {1: "webauthn.io", 2: h'AAC5C6515F67BCB340872496AF4ED3A932C8BFD5CEB194BD7C25BE5E7F45C136', 3: [{"id": h'473511DAAE38F6AA51ACCDE61986A4CFF6C7BDA6DAAADACABEAC12AA1E9FA04468703AD9EB66D9F23F3E195B9BFDC588', "type": "public-key"}, {"id": h'EBF0924BCE17BBD61C1092EC09E466A0ECFE91A89C30E9DD0C69FC019FA1429DB7D2756512AE9078498E237FC6B0395E', "type": "public-key"}], 6: h'AA9C95E78BB1D56D319AD7704FC157D9', 7: 1}
FIDODebug[08:22:56] -> {2: h'43A5576CB08E39A6CFC1B26BD3EA5676'}                 
FIDODebug[08:22:56] <- 6 {1: 1, 2: 5, 3: {1: 2, 3: -25, -1: 1, -2: h'6F7FF3D0AB7548E9454D8370704578635F879BA46D299940D2B2287405BAF427', -3: h'A473F3684B389D4A3235A08EA1D2AD712B86C1BEC3F787978D38D76C9D1D91E2'}, 6: h'6BCDF1A5BDB3DCE0A8073120A3E1CDFD'}
FIDODebug[08:22:56] -> {1: {1: 2, 3: -25, -1: 1, -2: h'419BBE453BFF6E89D69EF9D07F603FBA9BDFC71C25EA29D248340A264B51E942', -3: h'5856C3C8CD9B89D7970BE9A7F2251B0FEF40E9D9EA2ACBD019458386996A4E79'}}
FIDODebug[08:22:56] <- 6 {1: 1, 2: 2}                                           
FIDODebug[08:22:54] -> {3: 8}                                                   
FIDODebug[08:22:54] Ignoring status 45 from hid:1e7504ae-180b-45e6-8edb-e8a6f280af47
FIDODebug[08:22:54] -> (CTAP2 error code 45)                                    
FIDODebug[08:22:54] <- 6 {1: 1, 2: 1}                                           
FIDODebug[08:22:54] <- (cancel)                                                 
FIDODebug[08:22:54] -> (CTAP2 error code 51)                                    
FIDODebug[08:22:52] <- 1 {1: h'E3B0C44298FC1C149AFBF4C8996FB92427AE41E4649B934CA495991B7852B855', 2: {"id": ".dummy"}, 3: {"id": h'01', "name": "dummy"}, 4: [{"alg": -7, "type": "public-key"}], 8: h'', 9: 1}
FIDODebug[08:22:52] The device supports the CTAP2 protocol.                     
FIDODebug[08:22:52] Unexpected protocol version received.                       
FIDODebug[08:22:52] -> {1: ["U2F_V2", "FIDO_2_0", "FIDO_2_1_PRE"], 2: ["credProtect", "hmac-secret"], 3: h'79D699DF01914B10B9035467E7CE8231', 4: {"rk": true, "up": true, "plat": false, "credMgmt": true, "clientPin": true}, 5: 1200, 6: [1], 7: 20, 8: 256}
FIDODebug[08:22:52] <- 1 {1: h'E3B0C44298FC1C149AFBF4C8996FB92427AE41E4649B934CA495991B7852B855', 2: {"id": ".dummy"}, 3: {"id": h'01', "name": "dummy"}, 4: [{"alg": -7, "type": "public-key"}], 8: h'', 9: 1}
FIDODebug[08:22:52] The device supports the CTAP2 protocol.                     
FIDODebug[08:22:52] Unexpected protocol version received.                       
FIDODebug[08:22:52] -> {1: ["U2F_V2", "FIDO_2_0", "FIDO_2_1_PRE"], 2: ["credProtect", "hmac-secret"], 3: h'8976631BD4A0427F57730EC71C9E0279', 4: {"rk": true, "up": true, "plat": false, "credMgmt": true, "clientPin": true}, 5: 1200, 6: [1], 7: 20, 8: 128}
FIDODebug[08:22:52] Sending CTAP2 AuthenticatorGetInfo request to authenticator.
FIDOEvent[08:22:52] Starting GetAssertion flow                                  

```

as you can see, the same id was returned both times (the first one from the allowCredentials)

szszszsz, 2021-12-06:
Just to clarify, did you have both inserted at the time of the test?

flumm, 2021-12-06:
i believe during that test yes, but the result is the same even if i only have one plugged in at the time

flumm, 2021-12-09:
so, any ideas how to debug that? sadly i do not have a solo key hacker edition...
just fyi, this should also happen with 2 solokeys, since the onlykey uses part of the solokey firmware AFAIK

flumm, 2021-12-23:
is there anything i can do from my side? i'd really like to use my solokey as an additional factor to my onlykey...