solokeys/solo1 #592: Unprogrammed key?

noteknix, 2021-12-22:
I have 2 Solo-A keys both do not work. The first key was from the original kickstarter and was working, then seemed to stop. The latter only arrived today and as with the first one, Windows simply sees it as STM32Bootloader. 

Try using it as 2FA with Google, not detected, try using DiceKeys on a Mac, no device detected.   The original key did still go into program mode with the button pressed on insertion, that's now stopped, so it's possible I've bricked it or changed it to a Secure version, either way, Win11 just says STM32 Bootloader in device manager with either key inserted. The newer key does nothing when inserted with the button depressed or otherwise.  Solo Python is installed and detects neither key when using solo ls, just says ::SOLOS.   Wink doesn't work either.

msiqx9, 2021-12-27:
Seems you have un-programmed keys; a common problem.  I have had success in programming these blank keys using previously posted [instructions](https://github.com/solokeys/solo/issues/562#issuecomment-874014447).

noteknix, 2021-12-29:
With a bit of research, I now have signs of life in both keys.  Plug in and I get a pulsing green LED, press the button, blue, but that's my lot.  Solo ls gives me 205937A0324B: Solo 4.1.5 so I seem to have firmware? But insert the key with the button pressed I still get a pulsing green LED and any solo command says no DFU found.  Suggestions?

jolo1581, 2021-12-29:
Did you use solo phython?

Plug in solo and type 
```
solo ls
```

This should list all connected solo.

jolo1581, 2021-12-29:
Maybe you should try to Update your solo.

noteknix, 2021-12-29:
Yes, now returns the USB device ID and Fido2 device.

Which puts me sort of back to square 1, but at least both now have a steady green LED.

jolo1581, 2021-12-29:
Maybe a Win 11 problem?
Do you have any other system to test the keys?

Win 10 or Linux?

noteknix, 2021-12-29:
Mac, Win10, Win11 and Linux. :-) 

The Mac with Solo installed also displays the same results.  Might be time to try on Linux.

noteknix, 2021-12-29:
More work and I have a pair of working keys, except neither are recognised by the Solokeys app. Conclusion would be I've managed to turn at least the original key into a secure key as opposed to a hacker key?

msiqx9, 2021-12-31:
So far I have had to programme 2 blank keys and they both appear as "Hacker" in the Solokeys Windows app even though I used the regular secure firmware.  Both work fine...