solokeys/solo1 #593: solokey sometimes hangs

xogium, 2021-12-23:
Hi,
I bought a solokey a few months ago, and everything seemed ok with it, except that sometimes, it appears to be hanging.

My solo came with firmware 4.1.2, and already hit this issue. Now it is on 4.1.5, and it still happens.

My OS is up to date archlinux. I configured sudo command to require both the user password and a press on the solo's button to work, and my ssh key asks for the PIN of the solo plus button press.

I cannot reproduce this on-demand, only at random will it hit this state, but once it did, there's no way but to unplug and replug to make it go away, for a few days, maybe hours.

I'm left with hanging sudo or ssh commands waiting for user presence to be confirmed until they time out. The device is still enumerated fine, there is nothing weird showing in the system logs. It could be my OS being a problem, but I suspect the solo itself is hanging.

Any help would be appreciated.

It is currently in that strange state for now, I left it like this on purpose in case you wanted to ask for additional info.

xogium, 2021-12-23:
I checked the led just now, and it appears to be solid green no matter what.
It doesn't get solid red when asking for authentification at all... Which makes me think the key doesn't react at all, not even to auth requests, not just to button press.

foursixnine, 2022-01-20:
I wonder if this is somehow being looked at. 

I can confirm that the key's firmware hangs, whether being connected for a long period of time to a running computer, or a computer in suspended state.

xogium, 2022-01-20:
Oh, wow ! I'm not alone !

More seriously, I've bought a solu hacker and a somu locked (as my backup key), and I want to investigate the issue that way. I've just got no idea where to begin, to be perfectly honest.

That said so far I've not managed to get my somu to hang, interestingly enough. It has been plugged on for several days without any problem so far, which makes me wonder if this couldn't be a faulty chip ?

Anyway, I'll keep an eye on it.

Next time it hangs I want to try and see if 'solo key ping' will fail also, or somehow unfreeze it, who knows ?

foursixnine, 2022-01-21:
> Oh, wow ! I'm not alone !

I'm glad that we aren't :)

> More seriously, I've bought a somu hacker and a somu locked (as my backup key), and I want to investigate the issue that way. I've just got no idea where to begin, to be perfectly honest.

I've gotten it to hang often when my key is plugged and machine comes and goes from suspension state (it its connected to  a docking station)

> That said so far I've not managed to get my somu to hang, interestingly enough. It has been plugged on for several days without any problem so far, which makes me wonder if this couldn't be a faulty chip ?

It's too early for me to have a good picture too... and haven't had the time for now, to really look into it

xogium, 2022-04-03:
Hi,
after more than 2 months, I can confirm this looks more and more like an actua hardware issue with my solo.

I have never unplugged the somu I got, and have been using it since jan 20. It never hung like my solo did only after a couple hours at worst and few days at best...

I mean, it could certainly be a firmware issue, but it weirds me out if it is, because somu never ran into it, and I've been using it for the past 2 months without ever taking it out of the usb port.

eisenwinter, 2022-09-19:
Well sometimes is an understatement, I got mine today and it needs up to 10 re-plugs sometimes, it will stay solid green most of the time and the button press does nothing, the nfc feature just works but when plugging it in it almost never works...