solokeys/solo1 #597: Can't use Solo V2 with Google

gavinbarron, 2022-01-13:
Trying to add a new Solo V2 to my keys for Google and it fails with this message:

OS: Windows 11 Pro (Build 10.0.22000)
Browser: Chrome 97.0.4692.71 (Same result in Edge Dev)

[image]

Is there any way to get better diagnostics to help troubleshoot this issue?

Side note, it appears to work but fails with Microsoft Accounts but registered successfully for Twitter and GitHub.

timcappalli, 2022-01-13:
Sorry it is unclear from your post since you mention both "for Google" and "with Microsoft Accounts". What is the relying party you're trying to register a credential on?

gavinbarron, 2022-01-13:
Google.

nickray, 2022-01-13:
Wrong repo. See also https://github.com/solokeys/solo2/discussions/96 and https://github.com/solokeys/fido-authenticator/pull/8.

closed 2022-01-13 by nickray