solokeys/solo1 #600: Google Chrome U2F API decommission

realdmitchell, 2022-02-05:
Dear Devs

Sorry to link a different company's link.

https://www.yubico.com/blog/google-chrome-u2f-api-decommission-what-the-change-means-for-your-users-and-how-to-prepare/

But does this have any impact on the solokeys? I have Product: Solo 4.0.0
Should users of Solokey do something?

Thanks for your reply.

Zanthed, 2022-02-05:
Solokeys has support for FIDO2 which is what is meant to be used with Webauthn. U2F is deprecated and old, hence why it's being removed from Webauthn.

The key should work fine as long as it can do FIDO2 properly which you can test in places like https://demo.yubico.com/webauthn-technical/registration (it just needs to report back a valid registration, doesn't need to report the device itself).

realdmitchell, 2022-02-05:
Thank you for the quick reply. Very happy to know.

closed 2022-02-05 by realdmitchell

nickray, 2022-02-07:
Another way of putting this is that U2F was never an actual standard, but was renamed to CTAP1 and as such subsumed into the CTAP standard (where CTAP2 is the newer, more complex, extended interface). Under the hood, a lot of clients still actually use CTAP1 and not CTAP2; this story is a bit murky and intransparent.