solokeys/solo1 #607: Can't login to google with solo1 as 2FA

solsticedhiver, 2022-03-14:
I was able to setup my solokey (v1) as security usb key, and as 2FA on my google account.

But when I try to login, I am asked to use the key/press the button, it seems ok at first.

But then google finally display a message

    A problem happened.
    Try to reuse your security key or confirm you identity in any other way

This happens both on firefox 98 or chromium 99.0.4844.51

I have done a similar setup on github or dropbox and this wokrs without any issue.

Zanthed, 2022-03-14:
Is this a Solo hacker?

solsticedhiver, 2022-03-14:
No, a regular solokey

solsticedhiver, 2022-03-14:
OK. **I found out why.**

There is a difference when I setup thesecurity in firefox and chromium.
In chromium, I have dialog, that asks to autorise google to check the make and model of the key.

If I choose pass (or refuse), then I cannot login later on the the solkey.

I don't get such a dialog prompt on firefox I think.

If I choose, autohorize (to get make and model), I got no problem to use the solokey as expected.

solsticedhiver, 2022-03-14:
I am about to close the issue, because it is not an issue with the key, I guess, but more a browser or GUI issue.

But I have like an opinion of somone and/or a dev. Somone more knowladable thanme ,in FIDO2 and all those subjects.

solsticedhiver, 2022-03-15:
For the record, the message is:

    Authorize this site to read information from your security key ?

    google.com wants to to see the make and model of your security key

    Skip / Authorize

It should be more obvious that, by declining, you will be unable to login later on; is it a chromiun issue then ?

closed 2022-03-15 by solsticedhiver

nickray, 2022-03-23:
I'm not really sure, but (I think?) for a while Firefox by default did a flavor of obfuscation of your "make + model". They may still be doing that, and Google may not like this.

solsticedhiver, 2022-03-23:
During my test, I tried both options (masquerading or not) as asked in firefox. I don't remember if it makes a difference.

The dialog in firefox, is only asking about masquerading the information given to the site, or not. But something is given, whatever your choice.

The issue is more, in chromium/chrome, where the browser is asking you to give some info or not. If you pass, and refuse to give anything (a.k.a an attestation), the key does not work after that.

It should be made obvious, somewhere that refusing cause problem. At the very least, in the google process to add the key, and in some documentation, i don't know where. In solokeys doc ??, in webauthn doc ? I don't know.

I would have like to know this before discovering it the "hard" way.

Note: it is ironic that it is the very own google browser that is misleading, and allow to pass on something required.