solokeys/solo1 #608: Not showing under WSL

gibait, 2022-03-17:
This is a very long shot but i need to try one more time before giving up.. My setup is kinda complicated:
WSL2 on Windows 10, `usbipd` to allow USB passthrough in WSL. My `lsusb` ouput is this:
```
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 014: ID 0483:a2ca STMicroelectronics Solo 4.1.5
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```
So i suppose that the solo key is recognized. However, `dmesg` says:
```
[13970.816467] vhci_hcd vhci_hcd.0: pdev(0) rhport(0) sockfd(3)
[13970.816472] vhci_hcd vhci_hcd.0: devid(65538) speed(2) speed_str(full-speed)
[13970.816521] vhci_hcd vhci_hcd.0: Device attached
[13971.093671] vhci_hcd: vhci_device speed not set
[13971.163326] usb 1-1: new full-speed USB device number 14 using vhci_hcd
[13971.243673] vhci_hcd: vhci_device speed not set
[13971.313287] usb 1-1: SetAddress Request (14) to port 0
[13971.350648] usb 1-1: New USB device found, idVendor=0483, idProduct=a2ca, bcdDevice= 1.00
[13971.350652] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[13971.350653] usb 1-1: Product: Solo 4.1.5
[13971.350655] usb 1-1: Manufacturer: SoloKeys
[13971.350656] usb 1-1: SerialNumber: 207331B9414E
[13971.356610] hid-generic 0003:0483:A2CA.0008: device has no listeners, quitting
```
And `udevadm monitor` shows some wicked output:
```
KERNEL[13971.350617] add      /devices/platform/vhci_hcd.0/usb1/1-1 (usb)
KERNEL[13971.353805] add      /devices/platform/vhci_hcd.0/usb1/1-1/1-1:1.0 (usb)
KERNEL[13971.356378] add      /devices/platform/vhci_hcd.0/usb1/1-1/1-1:1.0/0003:0483:A2CA.0008 (hid)
KERNEL[13971.356460] bind     /devices/platform/vhci_hcd.0/usb1/1-1/1-1:1.0 (usb)
KERNEL[13971.356527] bind     /devices/platform/vhci_hcd.0/usb1/1-1 (usb)
KERNEL[13978.143518] unbind   /devices/LNXSYSTM:00/LNXSYBUS:00/ACPI0004:00/VMBUS:00/140752cc-a518-11ec-ae77-983b8fe9f529 (vmbus)
KERNEL[13978.143612] remove   /devices/LNXSYSTM:00/LNXSYBUS:00/ACPI0004:00/VMBUS:00/140752cc-a518-11ec-ae77-983b8fe9f529 (vmbus)
KERNEL[13978.143626] remove   /channels/50 (channels)
KERNEL[13978.143636] unbind   /devices/LNXSYSTM:00/LNXSYBUS:00/ACPI0004:00/VMBUS:00/140752cb-a518-11ec-ae77-983b8fe9f529 (vmbus)
KERNEL[13978.143651] remove   /devices/LNXSYSTM:00/LNXSYBUS:00/ACPI0004:00/VMBUS:00/140752cb-a518-11ec-ae77-983b8fe9f529 (vmbus)
KERNEL[13978.143690] remove   /channels/49 (channels)
KERNEL[13978.783526] unbind   /devices/LNXSYSTM:00/LNXSYBUS:00/ACPI0004:00/VMBUS:00/140752e5-a518-11ec-ae77-983b8fe9f529 (vmbus)
KERNEL[13978.783588] unbind   /devices/LNXSYSTM:00/LNXSYBUS:00/ACPI0004:00/VMBUS:00/140752d9-a518-11ec-ae77-983b8fe9f529 (vmbus)
KERNEL[13978.783609] remove   /devices/LNXSYSTM:00/LNXSYBUS:00/ACPI0004:00/VMBUS:00/140752e5-a518-11ec-ae77-983b8fe9f529 (vmbus)
KERNEL[13978.783628] remove   /devices/LNXSYSTM:00/LNXSYBUS:00/ACPI0004:00/VMBUS:00/140752d9-a518-11ec-ae77-983b8fe9f529 (vmbus)
KERNEL[13978.783673] remove   /channels/58 (channels)
KERNEL[13978.783714] remove   /channels/54 (channels)
KERNEL[13978.783758] unbind   /devices/LNXSYSTM:00/LNXSYBUS:00/ACPI0004:00/VMBUS:00/140752e4-a518-11ec-ae77-983b8fe9f529 (vmbus)
KERNEL[13978.783776] remove   /devices/LNXSYSTM:00/LNXSYBUS:00/ACPI0004:00/VMBUS:00/140752e4-a518-11ec-ae77-983b8fe9f529 (vmbus)
KERNEL[13978.783794] remove   /channels/57 (channels)
KERNEL[13978.783812] unbind   /devices/LNXSYSTM:00/LNXSYBUS:00/ACPI0004:00/VMBUS:00/140752d8-a518-11ec-ae77-983b8fe9f529 (vmbus)
KERNEL[13978.783874] remove   /devices/LNXSYSTM:00/LNXSYBUS:00/ACPI0004:00/VMBUS:00/140752d8-a518-11ec-ae77-983b8fe9f529 (vmbus)
KERNEL[13978.783942] remove   /channels/53 (channels)
KERNEL[13978.784000] unbind   /devices/LNXSYSTM:00/LNXSYBUS:00/ACPI0004:00/VMBUS:00/140752df-a518-11ec-ae77-983b8fe9f529 (vmbus)
KERNEL[13978.784060] remove   /devices/LNXSYSTM:00/LNXSYBUS:00/ACPI0004:00/VMBUS:00/140752df-a518-11ec-ae77-983b8fe9f529 (vmbus)
KERNEL[13978.784121] remove   /channels/56 (channels)
KERNEL[13978.784167] unbind   /devices/LNXSYSTM:00/LNXSYBUS:00/ACPI0004:00/VMBUS:00/140752de-a518-11ec-ae77-983b8fe9f529 (vmbus)
KERNEL[13978.784213] remove   /devices/LNXSYSTM:00/LNXSYBUS:00/ACPI0004:00/VMBUS:00/140752de-a518-11ec-ae77-983b8fe9f529 (vmbus)
KERNEL[13978.784276] remove   /channels/55 (channels)
KERNEL[13978.784337] unbind   /devices/LNXSYSTM:00/LNXSYBUS:00/ACPI0004:00/VMBUS:00/140752d2-a518-11ec-ae77-983b8fe9f529 (vmbus)
KERNEL[13978.784388] remove   /devices/LNXSYSTM:00/LNXSYBUS:00/ACPI0004:00/VMBUS:00/140752d2-a518-11ec-ae77-983b8fe9f529 (vmbus)
KERNEL[13978.784422] remove   /channels/52 (channels)
KERNEL[13978.784432] unbind   /devices/LNXSYSTM:00/LNXSYBUS:00/ACPI0004:00/VMBUS:00/140752d1-a518-11ec-ae77-983b8fe9f529 (vmbus)
KERNEL[13978.784449] remove   /devices/LNXSYSTM:00/LNXSYBUS:00/ACPI0004:00/VMBUS:00/140752d1-a518-11ec-ae77-983b8fe9f529 (vmbus)
KERNEL[13978.784485] remove   /channels/51 (channels)
```
Udev rules were installed correctly
```
# Notify ModemManager this device should be ignored
ACTION!="add|change|move", GOTO="mm_usb_device_blacklist_end"
SUBSYSTEM!="usb", GOTO="mm_usb_device_blacklist_end"
ENV{DEVTYPE}!="usb_device",  GOTO="mm_usb_device_blacklist_end"

ATTRS{idVendor}=="0483", ATTRS{idProduct}=="a2ca", ENV{ID_MM_DEVICE_IGNORE}="1"

LABEL="mm_usb_device_blacklist_end"

# Solo bootloader + firmware access
SUBSYSTEM=="hidraw", ATTRS{idVendor}=="0483", ATTRS{idProduct}=="a2ca", MODE="0660", GROUP="plugdev"
SUBSYSTEM=="tty", ATTRS{idVendor}=="0483", ATTRS{idProduct}=="a2ca", MODE="0660", GROUP="plugdev"

# ST DFU access
SUBSYSTEM=="usb", ATTRS{idVendor}=="0483", ATTRS{idProduct}=="df11", MODE="0660", GROUP="plugdev"

# U2F Zero
SUBSYSTEM=="hidraw", ATTRS{idVendor}=="10c4", ATTRS{idProduct}=="8acf", MODE="0660", GROUP="plugdev"
```
Now, i know this is a a complicated mess but i don't have the necessary knowledge to understand the source of the problem, so i turn to you. Feel free to move this to discussions. Thanks

szszszsz, 2022-03-17:
Just passing through, but perhaps I could help.

1. Can you show logs from solo1? 
2. This sounds interesting to follow:
    > [13971.356610] hid-generic 0003:0483:A2CA.0008: device has no listeners, quitting

gibait, 2022-03-17:
> Just passing through, but perhaps I could help.
> 
>     1. Can you show logs from solo1?
> 
I'm sorry, how do i obtain logs from solo1?
>     2. This sounds interesting to follow:
>        > [13971.356610] hid-generic 0003:0483:A2CA.0008: device has no listeners, quitting
Is this a driver problem? I tried to recompile WSL kernel with HID/HIDRAW modules. 
```
giba@Thinkpad-T480:~/solo1-cli$ ls -al /sys/bus/usb/drivers/usbhid
total 0
drwxr-xr-x 2 root root    0 Mar 17 11:32 .
drwxr-xr-x 8 root root    0 Mar 17 11:32 ..
lrwxrwxrwx 1 root root    0 Mar 17 13:27 1-1:1.0 -> ../../../../devices/platform/vhci_hcd.0/usb1/1-1/1-1:1.0
--w------- 1 root root 4096 Mar 17 11:32 bind
lrwxrwxrwx 1 root root    0 Mar 17 11:32 module -> ../../../../module/usbhid
-rw-r--r-- 1 root root 4096 Mar 17 11:32 new_id
-rw-r--r-- 1 root root 4096 Mar 17 11:32 remove_id
--w------- 1 root root 4096 Mar 17 11:32 uevent
--w------- 1 root root 4096 Mar 17 11:32 unbind
```

closed 2022-03-19 by gibait