solokeys/solo1 #617: New solo key usb-c not recognized by login sysem on linux

noccio, 2022-08-20:
I bought a second key usb-c because the previous was broken. I have another key usb-a working fine.
I tried to add the new usb-c key to login system with the command

    pamu2fcfg >> ~/.config/Yubico/u2f_keys

as explained in the documentation but the key is not recognized by login system and the led remains green.
I tried to change the `u2fkeys` file in many way but the problem remains and only the older key is working.

When I insert the key I saw this message in dmesg

    [ 7873.489067] usb 1-1: new full-speed USB device number 12 using xhci_hcd
    [ 7873.630419] usb 1-1: New USB device found, idVendor=0483, idProduct=a2ca, bcdDevice= 1.00
    [ 7873.630432] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
    [ 7873.630435] usb 1-1: Product: Solo 4.1.5
    [ 7873.630437] usb 1-1: Manufacturer: SoloKeys
    [ 7873.630439] usb 1-1: SerialNumber: 207432793141
    [ 7873.633955] hid-generic 0003:0483:A2CA.0008: hiddev96,hidraw1: USB HID v1.11 Device [SoloKeys Solo 4.1.5] on usb-0000:00:15.0-1/input0

And if I try to use the command line program `solo1` I have this error

    > solo key version                                                                                                                              
 
    Traceback (most recent call last):
    File "/home/leandro/.local/bin/solo", line 5, in <module> from solo.cli import solo_cli
    File "/home/leandro/.local/lib/python3.10/site-packages/solo/__init__.py", line 15, in <module> from . import client, commands, dfu, helpers, operations
    File "/home/leandro/.local/lib/python3.10/site-packages/solo/client.py", line 15, in <module> from .devices import solo_v1
    File "/home/leandro/.local/lib/python3.10/site-packages/solo/devices/solo_v1.py", line 11, in <module> from fido2.ctap1 import CTAP1
    ImportError: cannot import name 'CTAP1' from 'fido2.ctap1' (/usr/lib/python3.10/site-packages/fido2/ctap1.py)

The key usb-c works normally with the browsers.

I am on manjaro stable on amd64

noccio, 2022-08-20:
Closed issue

closed 2022-08-20 by noccio

craigcomstock, 2022-08-21:
> Closed issue

@noccio I am seeing this same problem (with python CTAP1 from fido2.ctap1) and you have closed this issue without noting what you did to fix the issue.

craigcomstock, 2022-08-21:
maybe this one? https://github.com/solokeys/solo1-cli/issues/151, with an open pull request. :+1:

craigcomstock, 2022-08-21:
Yes, this PR: https://github.com/solokeys/solo1-cli/pull/152 seems to fix at least the initial problem and likely handles what I need: programming a hacker solokey. 🎉 I hope.

craigcomstock, 2022-08-21:
After adjusting the udev rules from `TAG+="uaccess"` to `GROUP="plugdev"` (I am using a VM so maybe no elogind/etc/logged in user?) I can do `solo1 ls` so think I am good now. Thanks for the tickets and such.

joelmaxuel, 2022-10-16:
> Yes, this PR: https://github.com/solokeys/solo1-cli/pull/152 seems to fix at least the initial problem

A reminder that [PyPi](https://pypi.org/project/solo1/) continues to offer the version that has the same initial problem. Visits to this issue and [solo-cli#151](https://github.com/solokeys/solo1-cli/issues/151) should alleviate once that get updated.