solokeys/solo1 #623: Project dead?

an0th3rhuman, 2022-10-22:
Hi , I found one of the backer in reddit mentioned that this project is dead.

Could anyone please confirm?

vaporwave9, 2022-10-25:
Yes it is dead as they moved to different hardware, no I am not officially speaking or related to the project.
https://github.com/solokeys/solo2
https://github.com/solokeys/solo2/tree/main/runners/lpc55/board
Please refer to their new key.

I am just disappointed as it is less attractive now for "hackers". Since the dev board costs went from 15€ to 55€.
However the change is not to anger users, they had to move to new hardware to have all the features accessible.

I understand their choice, even if it changes a lot of things for us. Btw you can just buy a 
[SoloKeys](https://solokeys.com/) [Solo 2](https://solo2.dev/) and [Nitrokey 3](https://www.nitrokey.com/news/2021/new-nitrokey-3-nfc-usb-c-rust-common-criteria-eal-6/).
Instead if you need merely an authenticator. Somehow the two projects partnered up.

msiqx9, 2022-10-25:
My Solo v1 keys continue to work without any issues so I'm not bothered if it is now dead.  I recently received a Solo v2...

AlixAbbasi, 2022-11-08:
This is very weird thing to do. Because they are still selling the Solokey v1 in their website: https://solokeys.com/products/solo-usb-a?variant=23528224391232

pranqster, 2022-11-09:
Yeah. It's a total drag. I bought a few solov1 keys. They're nice keys and they all still work, but I was hoping for continued firmware updates. The firmware on the v2 is also written in Rust, which would have been nice to have for the v1.

bliu13, 2024-12-21:
Is there already a form of this codebase? It's kinda insane to just abandon this device when there is a lot of tickets filed and a load or PRs open. It would be nice if there was a community fork so that everyone's enhancements and fixes could be made into a release for general use.