solokeys/solo1 #624: Compatibility with ESP-WROOM-32??

notAZMAT, 2022-11-10:
Hey,
I was looking at the specs of ESP-32, it has features that satisfy quite a bit of the requirements for solo1 hardware.
I was thinking if the firmware would work on it, any suggestions?

merlokk, 2022-11-10:
esp32-s2 ideally fits.
needs to add another target here
https://github.com/solokeys/solo1/tree/master/targets
but the security will be very low (if it can have a name - it is negative security) )

notAZMAT, 2022-11-11:
@merlokk I guess it has flash encryption too that might not be too secure, is it secure enough?

merlokk, 2022-11-11:
its not so secure

coelner, 2022-11-18:
@notAZMAT https://limitedresults.com/2019/11/pwn-the-esp32-forever-flash-encryption-and-sec-boot-keys-extraction/
and https://www.espressif.com/en/news/Security_Advisory_Concerning_Fault_Injection_and_eFuse_Protections