solokeys/solo1 #629: Solo Hacker Not Found with WSL

yasmineantille, 2023-03-09:
I recently purchased the Solo Hacker (USB C) and am facing issues while following the getting-started docs. I've tried looking into older issues with similar problems here in the repo, but none of the suggested solutions have helped me completely solve this. 

I am working on a Windows machine, so I have WSL (version 2) installed with Ubuntu 22.04.2 LTS, where I am trying to generate and deploy new code to the device. I attach the usb device with ```usbipd wsl attach --busid <busid>``` and make sure with ```lsusb``` on ubuntu that it is recognized: 

```
Bus 001 Device 004: ID 0483:a2ca STMicroelectronics Solo 4.1.5
``` 

However now, running the command ```solo1 key version``` or ```solo1 key update``` on ubuntu I get the message: 

```
No Solo found.
If you are on Linux, are your udev rules up to date?
```

Running ```solo1 ls``` on ubuntu gives me the following: 

```
:: Solos
````

Running the same command on Windows powershell with admin rights when the device is not attached to ubuntu: 

```
:: Solos
204C3790324B: Solo 4.1.5
```` 

(I have updated the key's firmware on Windows in powershell with admin rights, so at least that step has worked, but now I'd like to successfully deploy the firmware from the docs to change the LED colors). 

I set up the udev rules according to https://github.com/solokeys/solo1-cli/issues/92#issuecomment-699078969 and tried finding the USB device's name by 
1) Running ```ls -1 /dev > ~/before.txt```, 
2) plugging the solo key in and attaching it to ubuntu, and then
3) running ```ls -1 /dev > ~/after.txt``` again and comparing the difference with ```diff ~/before.txt ~/after.txt``` but no diff is returned.

So my question is which (additional) steps are necessary to recognize the device with the solo cli using wsl? (Or is there a way to make the firmware on ubuntu, copying over the binaries to windows and then deploying the code to the device from there as I'm able to run commands to the device there?)

FaFre, 2023-04-17:
@yasmineantille  Did you find a solution for this?

yasmineantille, 2023-04-18:
> @yasmineantille Did you find a solution for this?

Sadly not. I opted to change to a bootable ubuntu system instead of using WSL (or Windows in general).

farazkaleemmalik, 2023-05-25:
> Sadly not. I opted to change to a bootable ubuntu system instead of using WSL (or Windows in general).
based