solokeys/solo1 #637: NFC triggers SoloKeys website

mcmikemn, 2024-03-08:
This may not be an issue with the device or software - it may be an issue with me.

I just got a Solo 2C+ NFC, but when I hold it up to the back of my phone, the web browser launches and loads https://solokeys.com. It doesn't seem to do anything as a security key.

I have a Pixel7 running GrapheneOS.

Has anyone else encountered this? Am I using it wrong? Thanks for guidance.

msiqx9, 2024-03-14:
I am also getting this on my Solo v2.  It's just annoying advertising.

mcmikemn, 2024-03-28:
> It's just annoying advertising that has been inserted into the key's firmware.

It's not just annoying advertising, it seems to have _replaced_ actual functionality.

kurtismiller01, 2025-02-17:
Same issue. I just got my solo 2C+ NFC and updated it. It works great until I need NFC to work on my S23 Ultra.